TerosTechnology/vscode-terosHDL #195: Per workspace configuration

suzizecat, 2021-07-22:
**Describe the bug**
On master, Teros configuration is not done through VSCode parameter but rather from a file in `~/.prj_config.json` .
This bring two issues :

 - There is a need for a per-workspace configuration on all parameters. 
    In example, in SystemVerilog, you need to specify include paths to lookup svh files.
    In VHDL that would be the language version specification. 
 - You can't have two instances of Teros running at the same time (I often have two or three of those)

khoitd1997, 2022-02-16:
Are there any plans to implement tackle this enhancement in the near future? Teros functionalities are awesome currently but the configurations file location can definitely be more flexible

qarlosalberto, 2022-02-16:
Not in a near future.

suzizecat, 2022-02-16:
Out of curiosity, which modification should be required in order to implement this ?

Wouldn't be possible to detect if a configuration file (I don't remember the exact name) is present in the root of the workspace and, if there is, read this one instead of the "global" one (located in the user directory) ?

suzizecat, 2022-06-02:
I'm coming back on this to add a small ping.

I am currently working on 4 different projects at the same time on the same machine and the fact that the configuration is virtually lost each time I switch a workspace is a huge bummer.

Not a show stopper, but quite the loss of time.

It is, IMHO, mandatory to somehow link :

- Settings with projects (you might not need the same setup for all projects)
- Projects with workspace (when I open a workspace, I don't care about the project that was open in the previous workspace but would love to get back the project(s) from my new workspace) 

As you mentioned working on a kind of overhaul, I'm willing to gives as many input as you'd wish in order to implement this.

A simple way to do so would be to store project and Teros settings in a path relative to the workspace root (maybe VSCode even have a path to do so) and load it each time a workspace is loaded. That shouldn't be too costly and would be a great user experience improvement. 

Would you mind giving me a pointer on where in teros codebase the settings are loaded ? I could have a go at implementing the feature.

qarlosalberto, 2022-06-02:
You completely are right. But currently I have stopped to add new features, I'm working in a complete back-end refactoring. And that occupies my free time totally.
First all it's necessary to design how it would be.

There are 2 kinds of configurations:

1. Global: editor, documenter, linter...
2. Per project: tool configuration.

I think that the best option is to save the "global options" as currently an to save "the project options" with each project. And when the user select a project TerosHDL will load the project options in the configuration menu.
So when you change of project you will change the configuration.

What do you think?

qarlosalberto, 2022-07-01:
Yes, it will include the linter for the project manager. Could you share a commands example for verilator... ? So it's easier for me to add it.

About the configuration for each project. Could you explain detailed how the user configure the project and general configuration? For example: the user clicks in the project and TerosHDL opens a GUI for the configuration... etc..
What happen if the user want to config it globally? Do I need to add a checkbox for a global config?

I know what is your point, but it's important to define the details. I need your feedback :)

suzizecat, 2022-07-02:
> I know what is your point, but it's important to define the details. I need your feedback :)

Please do not worry, i'd rather have an extensive discussion and get this done right 😉 

> Yes, it will include the linter for the project manager. Could you share a commands example for verilator... ? So it's easier for me to add it.

IIRC, the command is simply `verilator --lint-only -I/home/someone/included1 -Isubfolder/inc2  file1.sv file2.sv my_current_file.sv`
Where it is not even mandatory (but preferred) to mention directly the files.

The ideal would be to mention all project `.sv` files directly, and allows user to add as much included dirs as required, without adding them automatically. This can be left to the user to be done manually. (You could also have a tick somewhere to automatically add all project paths as included dir, but let the user a way to specifically add some custom paths.). Having arbitrary command line arguments is also a good backup (as long as the setting is project wise)

> About the configuration for each project. Could you explain detailed how the user configure the project and general configuration? For example: the user clicks in the project and TerosHDL opens a [...]

So, this one is a bit trickier.
To be fair, I do like the way VSCode handles it natively: you have a tab for global parameters and one for workspace.
When you watch the global parameters, you have an icon letting you know that "this" parameter is currently overriden by a workspace setting. Finally, you have an icon that tells you that a setting has been changed with regard to the default value and a button to revert back to the default value.

The default value means: 
 - For a project-wise configuration, it is the (current) global value of the parameter
 - For a global-wise configuration, it is the default value of the parameter, as hardcoded by you.

Be careful that the default value "nothing" might be different from "empty string" (I would suggest using string of length 0 as "nothing" (as in `None` in python) where appropriate while treating "blank filled" string as empty string and therefore equals to `""`), again, this is "when appropriate". The "reset" button would allow the user to ensure having `None` instead of `""`. 

For an interface (UI) point of view, I would probably be a lazy one an use VSCode settings as much as possible since they implement everything needed. 
Please note that it seems that you can now provide a custom order for those settings through the order property and handle arrays of strings (for arguments, in example)  : https://code.visualstudio.com/api/references/contribution-points#Configuration-schema

If you wish to, I can try a bit and do a dummy extension that would have Teros setting structure handled by vscode.

If, for whatever reason, you do not wish to use VSCode settings, I would start by arguing a bit about the added workload, then I would say that you can keep the current UI and "simply™️" add a new tab with the same settings, which would be the workspace settings. 
In example, you'd have "Global settings" in the yellow part and "Project <projectname> settings" in the blue one, on the screen capture bellow :
[image]

qarlosalberto, 2022-07-04:
Personally I don't like to have a large configuration (with a lot of items) in VSCode configuration. I think that it's a pain for the user. And the package.json is unreadable.  

But I like a lot your propose of Global/Project configuration. I think that the more easier would be to compare the project configuration with the default global configuration. If a filed is different then the user has modified the global configuration.
And also I could add a button for each field to reset it.

smgl9, 2022-07-04:
Additionally a couple of buttons to export/ import configurations would be nice. I think it can be useful to share configurations between developers. 
In fact it could be more than one file, including headers, style configuration, etc

suzizecat, 2022-07-05:
> But I like a lot your propose of Global/Project configuration. I think that the more easier would be to compare the project configuration with the default global configuration. If a filed is [...]

That the spirit ! :wink: 

> Personally I don't like to have a large configuration (with a lot of items) in VSCode configuration. I think that it's a pain for the user. And the package.json is unreadable.

I don't really like it either, but the point is that it holds all required features and saves you the implementation and testing.
As the user shouldn't spend much time in the settings anyway (once done globally for tools and/or once per project), I (but that's me) wouldn't have bothered using something else. 

I don't really mind though, as long as the features are here, I don't really mind the UI.

ecstrema, 2025-01-14:
Hey! I'm running into the same problem, but from a different angle: I configured the formatter for my vhdl code on my computer a few months back and formatted all the project gradually with it.

Now, I just changed computer, and since this configuration was global, I lost it in the computer change. Especially with formatter settings, I'd like to be able to save this to a specific workspace.

ecstrema, 2025-01-14:
> Personally I don't like to have a large configuration (with a lot of items) in VSCode configuration. I think that it's a pain for the user. And the package.json is unreadable.

Even though you are right that the ton of items in VSCode's settings is overwhelming, it is not any harder to click on the filter in the left of the settings than to open the terosHDL toolbar and open the global Teros settings.

[image]

This above shows the settings tree for the C++ extension. It is very similar to TerosHDL's own settings tree.

Migrating to this scheme would also allow you to remove quite a bit of code from this repository, which would simplify the maintenance and would give you more time for other more important things.