TerosTechnology/vscode-terosHDL #204: Support Chisel HDL

carlosedp, 2021-09-08:
This issue is a discussion thread and followup about adding Chisel support to TerosHDL extension.

Here I list the initial points I identify as requirement for support:

- Chisel is a DSL based on Scala so it has some pre-reqs:
	- Java JDK installed and configured (JAVA_HOME and PATH);
	- Some Scala tools might be needed like SBT or Mill;
	- If user wants to create a new project based on a template, [giter8](https://github.com/foundweekends/giter8) is advised;
	- Installation could be managed thru [Coursier](https://github.com/coursier/coursier). I add a reference below on how I setup my environment;
- Chisel projects can be built using SBT or Mill, depending on the project, either build file is supplied (example linked below);
- It would be great to include Scala and/or Chisel syntax highlighers, maybe more experienced Chisel devs could chime-in with recommendations for one or another;
- Chisel thru it's plugin generates the wires and regs in Verilog to match the variables in Chisel code, this could be used as click-thru or hover links between the sources for example;
- Chisel generates it's output as Verilog code, so most TerosHDL workflow could be used from this point on. The output dir could be set in the build tool params;
- Chisel tests can also be configured to generate VCD files in a specific diretory output, this could be leveraged by TerosHDL for dynamic displaying these files;
- The code generating functionality could work in a similar manner to FuseSOC chisel-generator here: https://github.com/fusesoc/fusesoc-generators/blob/master/chisel.py
- This generator also supports running the full generating workflow on a Docker container, not requiring local installation of tools (Java, etc). The feature uses the FuseSOC Launcher script [from here](https://gist.github.com/carlosedp/c0e29d55e48309a48961f2e3939acfe9) and is run as `EDALIZE_LAUNCHER=$(realpath ../runme.py) fusesoc run --target=ulx3s_85 carlosedp:demo:pmod1`.

There is a simple Chisel template based on giter8 from official Chisel at <https://github.com/freechipsproject/chisel-template> and also I created a more complete one at <https://github.com/carlosedp/chisel-template.g8>.

I also have a simple Blinky projects where I try to use and demo some Chisel features listed above. It's on https://github.com/carlosedp/chisel-playground.

Local dev environment setup example: <https://github.com/carlosedp/dotfiles/blob/master/setup_development.sh>

ekiwi, 2021-09-08:
This sounds like a great idea!

>     It would be great to include Scala and/or Chisel syntax highlighers, maybe more experienced Chisel devs could chime-in with recommendations for one or another;

For VS Code users we normally recommend that they install the `Scala (Metals)` plugin which not only provides syntax highlighting but also code completion and inspection features. Maybe you could just rely on that being installed and mostly try to port the unique visualization features of terosHDL to be available to Chisel users.

For IntelliJ, EasySoC made some Chisel/firrtl plugins that provide similar functionality and that you would install alongside the Scala plugin: https://plugins.jetbrains.com/organizations/EasySoC

drom, 2021-09-08:
here is https://github.com/chipsalliance/tree-sitter-firrtl that can be used for FIRRTL side of this project

qarlosalberto, 2021-09-15:
@carlosedp @drom @ekiwi thanks!! Sorry for the delay.

I need to read about Chisel, I'm a really noob.

@carlosedp could you do a list about what features related to Chisel do you want in TerosHDL? How @ekiwi have said the syntax highlight, go to definition... is covered by ```Scala (Metals)``` extension.

Do you run the simulations with Chisel (own simulator)? Or you need to transform the Chisel files to Verilog and to run the simulation? 

@carlosedp I share the full documentation for the next TerosHDL version. So, you can see all the features and to decide where would be nice to add Chisel support: https://drive.google.com/file/d/1KmnixIHoCQvewbJHWp2dauc0Qlr8o99Q/view?usp=sharing

You can download and install this version from: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

carlosedp, 2021-09-15:
Nice, TerosHDL is pretty new to me too as I never used it before.
I'll look deeper into it and return :)