TerosTechnology/vscode-terosHDL #207: Cannot set the path of Modelsim verilog linter

oksyassine, 2021-09-14:
Could you please add this in the settings of Teroshdl › Linter › Verilog › Linter › Modelsim: Path, as the one made for Vhdl ?

qarlosalberto, 2021-09-14:
What is your operative system?

oksyassine, 2021-09-14:
windows but I'm using remote ssh on a linux system

oksyassine, 2021-09-14:
Thanks @qarlosalberto for your reply,
**vcom** is working perfectly for me as we provide the path for vhdl linter but I need to do the same for verilog and provide the full path to **vlog**

qarlosalberto, 2021-09-14:
Install the beta version: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest
Click in the help button to see the full documentation:

[image]

oksyassine, 2021-09-14:
Could I have access also to put arguments like on vhdl right ?

qarlosalberto, 2021-09-14:
For now, you can't put arguments. But try the new version and tell me if you need a feature.

oksyassine, 2021-09-14:
Thank you very much for your support, very appreciated, this new version looks better. and I think there is a config to put arguments for vlog too:
[image]

I want to ask you, how to use both VCS and spyglass here ?

qarlosalberto, 2021-09-14:
For now the arguments are **only** for the simulation. To use VSC and spyglass you need to add the files to the project manager. Currently the documentation isn't online. You need to click in the help button and it will open your navigator.

Also you can download this folder and open the file index.html: https://github.com/TerosTechnology/vscode-terosHDL/tree/master/resources/project_manager/help

In the documentation webpage go to **project manager** section.

qarlosalberto, 2021-09-14:
Do it work vcom and vlog in the beta version?

oksyassine, 2021-09-14:
Thanks for the clarification.
Yes It worked for me.
Thanks for this great tool, I will check later for other features, simulation and spyglass lint rule checking..
I will close that issue for now.
Have a nice day!

closed 2021-09-14 by oksyassine