TerosTechnology/vscode-terosHDL #212: FuseSoC Support [Enhancement Draft]

saw235, 2021-09-18:
Add support for FuseSoC HDL package manager. 

The TerosHDL projects should automatically recognize the .core files and create a project for each target.
Then populates the target project with all the dependent files.

olofk, 2021-09-30:
Would love this as well. Happy to assist from the FuseSoC side when I can

qarlosalberto, 2021-10-09:
It would be really nice. First I need to play with FuseSoC, so I can imagine how to do the graphical interface.

WisniewskiP, 2023-04-25:
I'm using FuseSoc to build our project for a few months it works great.
At this moment I'm searching for better IDE support. we are using VSCode for many different project with different technologies and your extension would be great.

I have suggestion regarding "loading" FuseSoc project.
I think you could use two steps approach:
1. run `fusesoc --setup` to generate EDAM files
2. read generated EDAM files

you could observer .core files to rerun setup when one of those changes.

Similar approach is used by cmake plugin that we are using for c++ projects.

this would help with building top projects, discovering sub-cores may need some extra effort.