TerosTechnology/vscode-terosHDL #216: Documenter only shows last instantiation in file

awolnikowski, 2021-09-21:
**Describe the bug**
The "instantiations" section of the module documentation view only lists the last instantiation in any given Verilog file.

**To Reproduce**
Open the documentation preview for a Verilog module and scroll down to the instantiations section.

**Please complete the following information:**
 - OS: Ubuntu 18.04
 - VSCode version version 1.60.1

**Additional context**
I've only tried this on Verilog files. This seems like a pretty obvious bug but I was surprised to see no issues mentioning it.

qarlosalberto, 2021-09-21:
What TerosHDL version are you using?

awolnikowski, 2021-09-21:
I'm using version 0.1.4 of TerosHDL

qarlosalberto, 2021-09-21:
I think that this bug was fixed. We will release the new version this week or the next. Meanwhile please, use the beta version: https://github.com/TerosTechnology/vscode-terosHDL/releases

You can download the documentation from: https://drive.google.com/file/d/1KmnixIHoCQvewbJHWp2dauc0Qlr8o99Q/view?usp=sharing

awolnikowski, 2021-09-21:
Yep, looks great in the beta version. Thanks!

closed 2021-09-21 by awolnikowski

qarlosalberto, 2021-09-21:
Cool!