TerosTechnology/vscode-terosHDL #226: out put error

heartacker, 2021-09-29:
**Describe the bug**
```
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: ProtocolError("expected initialized notification, got Ok(Request(Request { id: RequestId(U64(1)), method: \"shutdown\", params: Null }))")', vhdl_ls\src\stdio_server.rs:89:14
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[Error - 下午2:03:55] Connection to server is erroring. Shutting down server.

```

**To Reproduce**
open code

**Code**
```
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: ProtocolError("expected initialized notification, got Ok(Request(Request { id: RequestId(U64(1)), method: \"shutdown\", params: Null }))")', vhdl_ls\src\stdio_server.rs:89:14
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[Error - 下午2:03:55] Connection to server is erroring. Shutting down server.

```

**Please complete the following information:**
 - OS: win10
 - VSCode version version 1.60.2

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.

qarlosalberto, 2021-09-29:
Does it happen all the time?

suzizecat, 2021-09-29:
@qarlosalberto If I may, do mind that there seems to be some non-ascii characters going on, that might trouble the language server at some point (?)

closed 2023-03-15 by qarlosalberto