TerosTechnology/vscode-terosHDL #227: Signal autocomplete

Pedroo64, 2021-09-29:
Hi,

I've upgraded terosHDL from 0.1.4 to 2.0.0 and while playing around with the extension, I noticed that signal auto complete is not working. Do I need to check something in the configurations to retrieve it?

Thanks
[image]
[image]

qarlosalberto, 2021-09-29:
What is your OS? 
Do the other features work? Eg: template generator.
Are you using non ASCII characters I'm the code?
Could you check in the output console? Have you this error?: https://github.com/TerosTechnology/vscode-terosHDL/issues/226

Pedroo64, 2021-09-29:
1. I'm on Windows (sorry I forgot to specify)
2. Yes, if I type "if" if block  template is generated
3. Yes I can go to signals definitions and navigate through components
4. I think so. At the bottom right it says "UTF-8"
5. I don't have any output... but a TerosHDL project has been created
[image]

qarlosalberto, 2021-09-29:
Could you install and test this beta version? https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest
Tell me if it fixes the bug.

Pedroo64, 2021-09-29:
Nice it's working thank you for your quick patch.
Still there is another bug (I think I should open another issue) that isn't really annoying but if you try to go to definition in the library **numeric_std** the linter throws this error.
[image]
And last, do the extension support also vhdl records autocomplete? Like in the example below, it should propose only the signals listed in the record. Of course **XtrRsp** is a type of **XtrRsp_t**.
[image]
[image]
Tell me if I should list this in different issues.
Thank you!
Btw this extension is awesome I've been using since last year I think and I recommended my students to use it! ^^

qarlosalberto, 2021-10-04:
Thanks for recommend it to your students! :)

numeric_std bug should be fixed in the latest beta version (https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest).

Currently it doesn't support records autocomplete, you can open a new issue for that.

closed 2021-10-05 by Pedroo64