TerosTechnology/vscode-terosHDL #234: Markdown export of documentation incorrect

GorgeousWalrus, 2021-09-30:
**Describe the bug**
When exporting the documentation to markdown, the tables are sometimes not correctly formatted

**To Reproduce**
Open code below, open documentation, export to markdown
```
--! Test design

library ieee;
use ieee.std_logic_1164.all;
use ieee.math_real.all;
use ieee.numeric_std.all;

entity test is
  generic (
    DWIDTH_g : natural := 1 --! Test generic
  );
  port (
    clk : in std_logic; --! Clock
    rstn : in std_logic; --! Active low reset
    din_i : in std_logic_vector(DWIDTH_g -1 downto 0); --! Data in
    dout_o : out std_logic_vector(DWIDTH_g -1 downto 0) --! Data out
  );
end entity;

architecture rtl of fifo is
  constant POINTER_SIZE_c : natural := natural(ceil(log2(real(DEPTH_g)))); --! Size of the write pointer (log2 of depth)
  constant FULL_c         : integer := (DEPTH_g);                          --! FIFO end
  constant EMPTY_c        : integer := 0;                                  --! FIFO start
  constant ALMOST_FULL_c  : integer := (DEPTH_g - AF_DISTANCE_g);          --! FIFO almost full
  constant ALMOST_EMPTY_c : integer := (AE_DISTANCE_g);                    --! FIFO almost empty

  signal write_pointer_s : std_logic_vector(POINTER_SIZE_c downto 0); --! Pointer of write address
  signal empty_s         : std_logic;                                 --! internal empty signal
  signal full_s          : std_logic;                                 --! internal full signal
  signal almost_empty_s  : std_logic;                                 --! internal almost empty signal
  signal almost_full_s   : std_logic;                                 --! internal almost full signal

  type t_fifo_mem is array (0 to DEPTH_g) of std_logic_vector(DWIDTH_g - 1 downto 0); --! FIFO memory type
  signal fifo_mem : t_fifo_mem;                                                       --! FIFO memory
begin
end architecture;
```

In addition to missing comments in the documentation (see [Issue 83](https://github.com/TerosTechnology/vscode-terosHDL/issues/83) ), the produced markdown code for some tables is badly formatted:

Produced Markdown (only tables):
```
| Generic name | Type    | Value | Description    |
| ------------ | ------- | ----- | -------------- |
| DWIDTH_g     | natural | 1     |  Test generic
 |
## Ports

| Port name | Direction | Type                                   | Description       |
| --------- | --------- | -------------------------------------- | ----------------- |
| clk       | in        | std_logic                              |  Clock            |
| rstn      | in        | std_logic                              |  Active low reset |
| din_i     | in        | std_logic_vector(DWIDTH_g -1 downto 0) |  Data in          |
| dout_o    | out       | std_logic_vector(DWIDTH_g -1 downto 0) |  Data out         |
## Signals

| Name            | Type                                      | Description                    |
| --------------- | ----------------------------------------- | ------------------------------ |
| write_pointer_s | std_logic_vector(POINTER_SIZE_c downto 0) |  Pointer of write address
     |
| empty_s         | std_logic                                 |  internal empty signal
        |
| full_s          | std_logic                                 |  internal full signal
         |
| almost_empty_s  | std_logic                                 |  internal almost empty signal
 |
| almost_full_s   | std_logic                                 |  internal almost full signal
  |
| fifo_mem        | t_fifo_mem                                |                                |
## Constants

| Name           | Type    | Value                               | Description                                 |
| -------------- | ------- | ----------------------------------- | ------------------------------------------- |
| POINTER_SIZE_c | natural |  natural(ceil(log2(real(DEPTH_g)))) |  Size of the write pointer (log2 of depth)
 |
| FULL_c         | integer |  (DEPTH_g)                          |  FIFO end
                                  |
| EMPTY_c        | integer |  0                                  |  FIFO start
                                |
| ALMOST_FULL_c  | integer |  (DEPTH_g - AF_DISTANCE_g)          |  FIFO almost full
                          |
| ALMOST_EMPTY_c | integer |  (AE_DISTANCE_g)                    |  FIFO almost empty
                         |
## Types

| Name       | Type | Description        |
| ---------- | ---- | ------------------ |
| t_fifo_mem |      |  FIFO memory type
 |
```

Markdown code in action:

| Generic name | Type    | Value | Description    |
| ------------ | ------- | ----- | -------------- |
| DWIDTH_g     | natural | 1     |  Test generic
 |
## Ports

| Port name | Direction | Type                                   | Description       |
| --------- | --------- | -------------------------------------- | ----------------- |
| clk       | in        | std_logic                              |  Clock            |
| rstn      | in        | std_logic                              |  Active low reset |
| din_i     | in        | std_logic_vector(DWIDTH_g -1 downto 0) |  Data in          |
| dout_o    | out       | std_logic_vector(DWIDTH_g -1 downto 0) |  Data out         |
## Signals

| Name            | Type                                      | Description                    |
| --------------- | ----------------------------------------- | ------------------------------ |
| write_pointer_s | std_logic_vector(POINTER_SIZE_c downto 0) |  Pointer of write address
     |
| empty_s         | std_logic                                 |  internal empty signal
        |
| full_s          | std_logic                                 |  internal full signal
         |
| almost_empty_s  | std_logic                                 |  internal almost empty signal
 |
| almost_full_s   | std_logic                                 |  internal almost full signal
  |
| fifo_mem        | t_fifo_mem                                |                                |
## Constants

| Name           | Type    | Value                               | Description                                 |
| -------------- | ------- | ----------------------------------- | ------------------------------------------- |
| POINTER_SIZE_c | natural |  natural(ceil(log2(real(DEPTH_g)))) |  Size of the write pointer (log2 of depth)
 |
| FULL_c         | integer |  (DEPTH_g)                          |  FIFO end
                                  |
| EMPTY_c        | integer |  0                                  |  FIFO start
                                |
| ALMOST_FULL_c  | integer |  (DEPTH_g - AF_DISTANCE_g)          |  FIFO almost full
                          |
| ALMOST_EMPTY_c | integer |  (AE_DISTANCE_g)                    |  FIFO almost empty
                         |
## Types

| Name       | Type | Description        |
| ---------- | ---- | ------------------ |
| t_fifo_mem |      |  FIFO memory type
 |

**Please complete the following information:**
 - OS: [Windows 10]
 - VSCode version version [1.57.1]

qarlosalberto, 2021-10-03:
I can't reproduce the error. Could you share your code as a file? Maybe there is a problem with the break lines, but it disappears when you copy the code here.

GorgeousWalrus, 2021-10-06:
On my machine, if I copy the code from here into a new file, the error is reproduced. Attached, you can find both the VHDL (code) and the generated markdown file

[test.zip](https://github.com/TerosTechnology/vscode-terosHDL/files/7292024/test.zip)

qarlosalberto, 2021-10-06:
It was a problem with the break line symbol in Windows: '\r'. Try with the new beta version: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

closed 2021-10-11 by GorgeousWalrus