TerosTechnology/vscode-terosHDL #236: Workspace full of errors with VHDL LS

am9417, 2021-09-30:
After upgrading to newest teros HDL version, my workspace is full of errors when opening any VHDL file, because declarations etc. in different files are not found. I don't want to configure TerosHDL project every time in the workspace but just to use the language support features (colouring, auto completion etc.) of the extension.

[image]

Tried to change VHDL linter in the configuration from VHDL LS to another but no effect.

wimille, 2021-09-30:
Agree with @am9417, TerosHDL should given the possibility to disable the linting features, which can be very usefull in some cases, but not all the time.

qarlosalberto, 2021-10-05:
I will take a look. Meanwhile you can configure the linter for only the current file:

[image]

qarlosalberto, 2021-10-09:
It should be fixed in the new version.

wimille, 2021-10-11:
Hi,
i've updated terosHDL, and now no warnings appear in vscode. However i've do nothing more than just update the extension. But in the changelog i've saw the "Added ability to disable VHDL LS linter." but i don't see this option ? Is i miss something ?