TerosTechnology/vscode-terosHDL #237: Parameter are not passed to Verible Linter

suzizecat, 2021-10-03:
**Describe the bug**
Parameters sets in the "VeribleLint" section of Teros configuration are not passed to the command line used to actually lint when using Verible as a Style Linter.

I have these parameters: 
[image]

[image]

And this message upon linting : 

```text
[colibri][info] Linting with command: verible-verilog-lint     /home/julien/Projets/HDL/Hamster/peripherals/spi/rtl/spi_master.sv
```

The tool indeed keeps reporting tabs everywhere, so the parameters are not taken into account.

**To Reproduce**
Use verible as a style linter on a code that violate any of its rules, then try to deactivate the rule through command line arguments.

**Code**

**Please complete the following information:**
 - OS: Ubuntu 20.04
 - VSCode version: 
 ```text
Version: 1.60.2
Commit: 7f6ab5485bbc008386c4386d08766667e155244e
Date: 2021-09-22T12:01:43.795Z
Electron: 13.1.8
Chrome: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Linux x64 5.4.0-88-generic snap
```

qarlosalberto, 2021-10-04:
Try with the new beta version.

qarlosalberto, 2021-10-06:
Could I close the issue?