TerosTechnology/vscode-terosHDL #240: Python virtualenv does not work

Hardolaf, 2021-10-04:
**Describe the bug**
[Per the documentation](https://terostechnology.github.io/terosHDLdoc/configuration/python.html), a Python3 virtualenv should be supported. I attempted to use this with an existing virtualenv but terosHDL appears to fail to activate it.

**Please complete the following information:**
 - OS: RHEL 7.9
 - VSCode version: 1.60.2
 - Python version: 3.8.5

**Log**
```
****************************************************************************************************************************************
---> Python3 path: doesn't detected.
---> Python3 packages directory: 
---> VUnit is NOT installed.
---> Cocotb is NOT installed.
---> yowasp-yosys is NOT installed.
---> Edalize is NOT installed.
---> Make is installed.
****************************************************************************************************************************************
Configure your Python 3 path correctly.
Install VUnit, yowasp-yosys, Edalize manually or install teroshdl python libraries: pip install teroshdl
Check the documenatation: https://terostechnology.github.io/terosHDLdoc/about/requirements.html
****************************************************************************************************************************************
2021-10-04 09:14:10: Error Python3 path (https://terostechnology.github.io/terosHDLdoc/configuration/python.html). Current Python3 path: source ~/workspace/venv/bin/activate; python
```

qarlosalberto, 2021-10-04:
Now it should work. Try with the beta version: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest
And check the configuration again: https://terostechnology.github.io/terosHDLdoc/configuration/python.html#example-with-virtualenv

```
. .venv/bin/activate; python3
```

Hardolaf, 2021-10-04:
This is working and loading the virtual environment. However, it's not able to find the packages installed inside of the virtualenv.

*Verify Output*
****************************************************************************************************************************************
---> Python3 path: /home/hardolaf/workspace/venv/bin/python3
---> Python3 packages directory: /censored/python38/lib/python38.zip,/censored/python38/lib/python3.8,/censored/python38/lib/python3.8/lib-dynload
---> VUnit is NOT installed.
---> Cocotb is NOT installed.
---> yowasp-yosys is NOT installed.
---> Edalize is NOT installed.
---> Make is installed.
```
****************************************************************************************************************************************
Install VUnit, yowasp-yosys, Edalize manually or install teroshdl python libraries: pip install teroshdl
Check the documenatation: https://terostechnology.github.io/terosHDLdoc/about/requirements.html
****************************************************************************************************************************************
```

*pip Output*
```
$ pip list
Package                    Version             Location
-------------------------- ------------------- ---------------------------------
anytree                    2.8.0
appdirs                    1.4.4
asn1crypto                 1.4.0
attrs                      21.2.0
azure-common               1.1.27
azure-storage-blob         2.1.0
azure-storage-common       2.1.0
bcrypt                     3.2.0
beautifulsoup4             4.9.3
boto3                      1.11.17
botocore                   1.14.17
certifi                    2020.12.5
cffi                       1.13.2
cfgv                       3.3.0
chardet                    3.0.4
colorama                   0.4.4
cryptography               2.9
cycler                     0.10.0
dataclasses                0.8
distlib                    0.3.1
docopt                     0.6.2
docutils                   0.15.2
dohq-artifactory           0.7.508
edalize                    0.2.5
filelock                   3.0.12
fusesoc                    1.12.0
future                     0.18.2
identify                   2.2.6
idna                       2.8
ijson                      2.6.1
importlib-metadata         4.2.0
importlib-resources        5.1.4
influxdb                   5.3.1
iniconfig                  1.1.1
ipyxact                    0.2.4
Jinja2                     3.0.1
jmespath                   0.10.0
jsonschema                 3.2.0
kafka-python               2.0.2
kiwisolver                 1.3.1
MarkupSafe                 2.0.1
matplotlib                 3.3.4
msgpack                    1.0.2
netmiko                    3.3.0
nodeenv                    1.6.0
numpy                      1.19.5
okonomiyaki                1.3.2
oscrypto                   1.2.1
packaging                  21.0
packetcowboy               5.1.267
pandas                     1.1.5
paramiko                   2.7.2
Pillow                     8.2.0
pip                        21.2.4
pluggy                     0.13.1
pre-commit                 2.13.0
prompt-toolkit             3.0.18
py                         1.10.0
pyaml                      20.4.0
pyarrow                    4.0.0
pycparser                  2.20
pycryptodomex              3.10.1
PyJWT                      1.7.1
PyNaCl                     1.4.0
pyOpenSSL                  19.1.0
pyparsing                  2.4.7
pyrsistent                 0.18.0
pyserial                   3.5
pysphere                   0.1.7
pyteamcity                 0.1.1
pytest                     6.2.4
python-dateutil            2.8.1
pytz                       2020.5
PyYAML                     5.4.1
rawutil                    2.7.0
requests                   2.22.0
s3transfer                 0.3.7
scapy                      2.4.4
scp                        0.13.3
setuptools                 40.6.2
simplesat                  0.8.2
six                        1.16.0
snowflake-connector-python 2.2.1
soupsieve                  2.2.1
tabulate                   0.8.9
teamcity-messages          1.28
teroshdl                   2.1.1
textfsm                    1.1.0
toml                       0.10.2
typing-extensions          3.10.0.0
urllib3                    1.25.11
virtualenv                 20.4.7
vunit                      0.1.3
vunit-hdl                  4.5.0
wasmtime                   0.28.1
wcwidth                    0.2.5
yowasp-yosys               0.9.post5610.dev226
zipfile2                   0.0.12
zipp                       3.4.1
```

qarlosalberto, 2021-10-04:
What is your installation path?

Hardolaf, 2021-10-04:
Ah sorry about the last report. My sysadmin updated Python in the path underneath me to 3.8 and I didn't update the virtualenv. This feature is working as expected now.

closed 2021-10-04 by Hardolaf