TerosTechnology/vscode-terosHDL #244: Teros diable word Autocomplete

Ahmad-Zaklouta, 2021-10-06:
**Describe the bug**
When I install Teros, the Autocomplete based on words in the file stopped working.
I tried to uninstall it and then Autocomplete worked again.
It seems with Terso you get autocomplete only for code. Not for comments for example or any string.

**To Reproduce**
Try to type Hello. and then type H, no suggestion for Hello.

**Please complete the following information:**
 - OS: Win 10
 - VSCode version[1.60.2]

qarlosalberto, 2021-10-08:
The result is as expected. TerosHDL provides autocomplete for VHDL, SV and Verilog, so autocomplete based on words is disabled by VSCode. It isn't possible to use both.

Ahmad-Zaklouta, 2021-11-01:
Is there an option to disable TerosHDL autocomplete and just use VSCode autocomplete? I don't need the VHDL autocomplete from TerosGDL

qarlosalberto, 2021-11-01:
No, but I could add it.

Ahmad-Zaklouta, 2021-11-09:
> No, but I could add it.

I think that would be great. I think many have there own snippets/keybinding and just need a word autocomplete

QuteSaltyFish, 2021-12-27:
You can use Tabnine to add word completion

PonakLiu, 2023-07-07:
i meet the same question, unfortunately I also unable to use its own autocomplete functions (It can prompt for variables but cannot prompt for keywords), i want to know how to disable dis feature or make it useful?