TerosTechnology/vscode-terosHDL #251: Add TCL language support

Nick-KV, 2021-10-08:
**Add support of TCL language**
As TCL is main language for creating/run various commands in any FPGA/ASIC environment it could be great to get full support of this script language.

**Highlight syntax**
As TCL pretty old script language it has a number syntax specifications. All of them could be find in any profile site.
Example: calling variable '$', ignoring next symbol '\', execute first in '[ ]' brackets, insert as is in '{ }' brackets, etc.
In additional - language support clever line breaking by '\' symbol, implementing could be really difficult.

**Code definitions and keywords**
Add basic keywords from any TCL.Wiki site (like this https://wiki.tcl-lang.org/)
In additional - get opportunity to add keywords for specific tool/EDA in some list-setting.

**Support of extended variable naming**
All variables in TCL are strings. Working and converting selected by operation type with variables
In general variables created by command 'set':
set my_variable "This is my string"
variables preferred to be "monolythic" with lover space or Camel-like naming. Anyway it is supporting to create names with whitespace:
set my\ variable "This is my string too" / set {my variable} "This string valid too"
As well variables could be 'arrays' (some values in table), 'lists' (just names/numbers with separator) and 'vectors' (number range).
Additionally variables could be 'associative arrays':
set my_var(main,first_worker) "Antonio Banderas"
It should be colored as regular variables ( $my_var(main,first_worker) ) and if possible highlighted by double klick as one word [didn't see this functional anywhere]

**Go to definition and Hover**
As TCL could call another files/scripts it would be great to get opportunity decide what project structure of TCL files. As some variables could be determined in another file with option default structure/project parsing.

**Template generator**
If possible - will be good. Especially with user option/editing function. Same could be used for user-determined 'functions', called 'proc'.

*** Other options could be added on author opinion ***

qarlosalberto, 2021-10-09:
Thanks for the large description. In the new version (v2.0.2) I have added an early support for TCL:

- Highlight syntax.
- "go to definition"/hover for the current file (not cross-file).
- Snippets.

The option to add keywords for specific tool/EDA would be very useful (I use Xilinx tcl). I need to think how implement it.

Feel free to suggest improvements to the v2.0.2 version.

qarlosalberto, 2023-06-26:
Dine

closed 2023-06-26 by qarlosalberto