TerosTechnology/vscode-terosHDL #255: VHDL-LS linter process "vhdl_ls" remains active when closing VSCodium

patrickrst, 2021-10-13:
Opening and closing VSCodium in repetition creates multiple processes:

```
asdf     11812  0.0  0.6 415172 54512 ?        Sl   10:49   0:00 /home/asdf/.vscode-oss/extensions/teros-technology.teroshdl-2.0.2/server/vhdl_ls/v0.1.8/vhdl_ls-x86_64-unknown-linux-gnu/bin/vhdl_ls enable_linter
asdf     12176  0.0  0.6 415040 53848 ?        Sl   10:51   0:00 /home/asdf/.vscode-oss/extensions/teros-technology.teroshdl-2.0.2/server/vhdl_ls/v0.1.8/vhdl_ls-x86_64-unknown-linux-gnu/bin/vhdl_ls enable_linter
asdf     20746  0.0  0.7 415040 57232 ?        Sl   14:46   0:00 /home/asdf/.vscode-oss/extensions/teros-technology.teroshdl-2.0.2/server/vhdl_ls/v0.1.8/vhdl_ls-x86_64-unknown-linux-gnu/bin/vhdl_ls enable_linter
asdf     23424  0.2  0.6 415044 52792 ?        Sl   15:43   0:00 /home/asdf/.vscode-oss/extensions/teros-technology.teroshdl-2.0.2/server/vhdl_ls/v0.1.8/vhdl_ls-x86_64-unknown-linux-gnu/bin/vhdl_ls enable_linter
...
```

 - OS: Debian 10
 - VSCodium version 1.61.0

qarlosalberto, 2021-10-16:
Thanks, I can reproduce it.

randre70, 2022-02-09:
Me too, also vhdl_ls is still running after vscode is closed. Keeps files locked and thumb drives can't be ejected.

qarlosalberto, 2023-06-26:
Fixed on v5.0.0

closed 2023-06-26 by qarlosalberto