TerosTechnology/vscode-terosHDL #256: [Feature request] Custom VHDL formatter

elpollodiablo84, 2021-10-15:
Since there is only one VHDL formatter bundled with Teros, it could be useful to add the option to execute a custom script/command/external program to format VHDL code (for example, I use a python library called [vhdl-style-guide]( https://github.com/jeremiah-c-leary/vhdl-style-guide)). I think a simple execution of the binary specified by the user with argument the current file should suffices.

Thanks for your precious work guys/gals!

qarlosalberto, 2021-10-16:
Check the beta version: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

qarlosalberto, 2021-10-16:
https://user-images.githubusercontent.com/5955143/137569284-7d867736-e9e2-436c-9a40-a32bf4a322e2.mp4

elpollodiablo84, 2021-10-18:
This is much more than what I expected, nice work!