TerosTechnology/vscode-terosHDL #267: TerosHDL doesn't find Yosys, GHDL or YoWASP to generate Schematic

FCPlech, 2021-11-05:
**Description**
I have installed GHDL and Yosys and using via terminal without no problems until now. Both are added on the system path. I also installed the sugested option (YoWASP), but had no success. The tool continues to show the message:

*Configure (https://terostechnology.github.io/terosHDLdoc/netlist/configuration.html) Yosys or install YoWASP: pip install yowasp-yosys* (If I select GHDL + Yosys)

*Your project/file includes 1 or more VHDL files, but it's not configured the backend GHDL+Yosys (https://terostechnology.github.io/terosHDLdoc/netlist/configuration.html).* (If I select YoWASP or only Yosys)

**Additional information:**
 - OS: Zorin OS 15.3
 - VSCode version: 1.60.0

**Screenshots**
[image]

qarlosalberto, 2021-11-05:
Check this issue: https://github.com/TerosTechnology/vscode-terosHDL/issues/247

qarlosalberto, 2021-11-11:
Could I close de issue?

closed 2021-11-16 by qarlosalberto