TerosTechnology/vscode-terosHDL #269: VHDL-LS Linter Not Working

Guzbikes, 2021-11-08:
I recently installed TerosHDL into VSCode 1.61.0 on a RedHat 7.9 server that is not connected to the internet, and I am not getting automatic linting using the VHDL-LS tool.

**To Reproduce**
Run TerosHDL in VSCode 1.61.0 on RedHat 7.9 and type some incorrect syntax

 - OS: RedHat 7.9
 - VSCode version version 1.61.0

**Additional context**
This system is not connected to the internet, but I'm not sure if that has anything to do with this issue. 

What debugging steps can I follow to see if the VHDL-LS server is running or not? Is there any additional setup I need to do, or any additional dependencies or packages I need to install for VHDL-LS, beyond installing the TerosHDL extension?

qarlosalberto, 2021-11-08:
I think that it doesn't work in RedHat 7.9. Anyway I will add an advice in the TerosHDL console if it detects an error to initialize VHDL-LS.

@suzizecat could you help him? You use RedHat, right?

Guzbikes, 2021-11-08:
> I think that it doesn't work in RedHat 7.9. Anyway I will add an advice in the TerosHDL console if it detects an error to initialize VHDL-LS.
> 
> @suzizecat could you help him? You use RedHat, right?

Thanks for the quick reply! Is it just VHDL_LS that doesn't work? Are there other linters that will work in this version of Redhat? I have questasim/modelsim installed on the same server but haven't tried it yet for linting.

suzizecat, 2021-11-09:
Hello @Guzbikes ,
The issue with RHEL7 is, on my side, that standard C library are too old to allow Teros backend to run.
You might want to go to `~/.vscode/extensions/teros-technology.teroshdl-2.0.4/server/vhdl_ls/v0.1.8/vhdl_ls-x86_64-unknown-linux-gnu/bin` (the path might change sligthly for you) and try to run `./vhdl_ls -h`.

On my side, I get `./vhdl_ls: /lib64/libc.so.6: version `GLIBC_2.18' not found (required by ./vhdl_ls)` which means that I am sad and without the greatness that Teros can achieve 😭 
Unfortunately, this is an issue that arise with most of the open source HDL tools (slang, verible, ghdl if I remember correctly)  under RH

You can, however use a professional tool such as modelsim, but to achieve the best result, you will want to statically pass the list of files of your design through Teros parameters, in order to avoid all the "missing modules" stuff. The performances and features provided through this solution are generally and globally worse than using vhdl_ls (in example, you will not have "go to definition" through files)  but it's better than nothing and can still catch a lot of issues.

Guzbikes, 2022-01-20:
I got VSCode up and running on a new RedHat 8.5 VM! So now I see the VHDL LS linter running and it is Linting stuff. But it identifies all library declarations as not found.

Is it possible to *just* do syntax checking, and turn off all file hierarchy and library checking? I tried unchecking the box in the TerosHDL configuration: "Activate go to definition feature for VHDL" but that turned off syntax checking too.

qarlosalberto, 2022-01-20:
Have you tried to add your libraries to the project manager?

Guzbikes, 2022-01-20:
I was hoping not to have to :-) But that does work for my local library files. So now it is showing my technology libraries as errors. And those are currently in Questa qdb format. Any ideas on how to get those to resolve or how to ignore them?

qarlosalberto, 2022-01-20:
Currently it's not possible :(

Could you send me an email to carlosruiznaranjo@gmail.com ?

Guzbikes, 2022-01-20:
Sure, I will email you from my personal account later today. In the meantime, couldn't I just create a vhdl file with the same architecture name just to remove the warnings?

qarlosalberto, 2022-01-20:
It's a very ugly way, but it could work.

Guzbikes, 2022-01-21:
Actually didn't even need to add a new file. I just clicked the "Add library" button in the project manager and created a library of the same name. That gets rid of the warnings!

qarlosalberto, 2022-01-21:
Cool! Please, send me the email, I would like to ask you a thing.

closed 2022-01-21 by qarlosalberto