TerosTechnology/vscode-terosHDL #270: Shared variable causes test failure

HuskyNator, 2021-11-10:
**Describe the bug**
When adding a shared variable to an architecture, an error is reported, causing test failure.
I'm fairly certain this is unintended behavior, as the warning should seemingly be inconsequential.

**To Reproduce**
Create the example GHDL project & add the line `shared variable c : bit;` below line 12 of the test bench.

**Code**
hald_adder_tb.vhd:
```
-- half_adder_process_tb.vhd

library ieee;
use ieee.std_logic_1164.all;
use std.env.finish;

entity half_adder_process_tb is
end half_adder_process_tb;

architecture tb of half_adder_process_tb is
    signal a, b : std_logic;
    signal sum, carry : std_logic;
    shared variable c: bit;

    component half_adder
        port (
        a : in std_logic;
        b : in std_logic;
        sum : out std_logic;
        carry : out std_logic
    );
    end component;

begin
    -- connecting testbench signals with half_adder.vhd
    UUT : half_adder port map (a => a, b => b, sum => sum, carry => carry);

    tb1 : process
        constant period: time := 20 ns;
        begin
            a <= '0';
            b <= '0';
            wait for period;
            assert ((sum = '0') and (carry = '0'))  -- expected output
            -- error will be reported if sum or carry is not 0
            report "test failed for input combination 00" severity failure;

            a <= '0';
            b <= '1';
            wait for period;
            assert ((sum = '1') and (carry = '0'))
            report "test failed for input combination 01" severity failure;

            a <= '1';
            b <= '0';
            wait for period;
            assert ((sum = '1') and (carry = '0'))
            report "test failed for input combination 10" severity failure;

            a <= '1';
            b <= '1';
            wait for period;
            assert ((sum = '0') and (carry = '1'))
            report "test failed for input combination 11" severity failure;

            finish;
        end process;
end tb;
```

**Please complete the following information:**
 - OS: Windows 10
 - VSCode version: 1.62.1

**Additional context**
Teminal output:
```
****************************************************************************************************************************************
2021-11-10 18:05:48: python "c:\Users\***\.vscode\extensions\teros-technology.teroshdl-2.0.4\resources\project_manager\run_edalize.py" "C:\Users\***\AppData\Local\Temp\edam.json" "ghdl" "C:\Program Files\GHDL\bin" "" "false" "vcdrom"
"Creating libraries directories"
ghdl -i --std=08 --ieee=synopsys  c:\Users\***\.vscode\extensions\teros-technology.teroshdl-2.0.4\resources\project_manager\examples\ghdl\half_adder_tb.vhd c:\Users\***\.vscode\extensions\teros-technology.teroshdl-2.0.4\resources\project_manager\examples\ghdl\half_adder.vhd
"Creating libraries directories"
ghdl -i --std=08 --ieee=synopsys  c:\Users\***\.vscode\extensions\teros-technology.teroshdl-2.0.4\resources\project_manager\examples\ghdl\half_adder_tb.vhd c:\Users\***\.vscode\extensions\teros-technology.teroshdl-2.0.4\resources\project_manager\examples\ghdl\half_adder.vhd
ghdl -m --std=08 --ieee=synopsys  half_adder_process_tb
c:\Users\***\.vscode\extensions\teros-technology.teroshdl-2.0.4\resources\project_manager\examples\ghdl\half_adder_tb.vhd:13:21: type of a shared variable must be a protected type
make: *** [Makefile:16: half_adder_process_tb] Error 1

************************************************************************************************
---> Build directory: C:\Users\***\.teroshdl\build
---> The test has finished with errors!
****************************************************************************************************************************************
```

qarlosalberto, 2021-11-10:
It's a GHDL issue, please ask in GHDL repository :)

closed 2021-11-10 by qarlosalberto

HuskyNator, 2021-11-10:
> The VHDL-2008 standard made it mandatory for all shared variables to be a protected type.
Hmm, seems to possibly be intended behavior.