TerosTechnology/vscode-terosHDL #271: Verible formatter doesn't work on Windows

PascalZh, 2021-11-12:
**Describe the bug**
Can't use verible formatter on Windows. The codes will not change when format is pressed.

**To Reproduce**
Set Verible formatter binary path: `C:\Users\Pascal\Programs\verible\verible-verilog-format.exe` and format with Verible formmater.

**Please complete the following information:**
 - OS: Windows 11
 - VSCode version version: 1.62.1

**Screenshots**
[image]

**Additional**
The executable is working well from PowerShell.

qarlosalberto, 2021-11-12:
Could you share the full command when you use it from PoweShell?

PascalZh, 2021-11-12:
> Could you share the full command when you use it from PoweShell?

[image]

PascalZh, 2021-11-12:
> Could you share the full command when you use it from PoweShell?

I find out it is the problem with Verible.
verible **will not format any lines** when any syntax error occurs, even if the syntax error occurs after the lines that should be reformatted.

Here are the syntax error messages from verible formatter:
[image]

This message is provided as warning from verible linter:
[image]

I suggest this message should be shown as errors.

qarlosalberto, 2021-11-12:
Yes good idea.

PascalZh, 2021-11-22:
> Yes good idea.

I find we can format the file anyway by turn fail fast option of verible-verilog-format off:
[image]

closed 2022-03-25 by PascalZh