TerosTechnology/vscode-terosHDL #272: Add suggestions to signals attributes and records

danieldsleal, 2021-11-12:
When trying to check the attributes for a signals TerosHDL doesn't show any possibilities.

For example, when writing " Signal_Array' " a list of suggestions should appear, like length, size, etc...

Also, when using record types, when writing "Signal_Array." a list of all possibilities of the record should appear.

How it works on ModernVHDL, for example:

[image]

danieldsleal, 2021-11-12:
Actually ModernVHDL doesn't seem to do what I told about the records, but it would be a nice addition.

qarlosalberto, 2022-01-12:
The list of suggestions (image, size...) will be add in the next version.