TerosTechnology/vscode-terosHDL #277: leaving state in fsm if begin end not used in verilog

cbiker, 2021-11-22:
**Describe the bug**
if code of case not use begin end in verilog state machine viewer can lost condition and transition to this state

**To Reproduce**
module pattern_generator_stm ( bla bla );
begin    // if module not used begin end fsm is not recognized   
    enum logic[3:0] {bla} state;
     //! fsm_extract
    always @(posedge clk )
    begin // if always not used begin end fsm is not recognized  
        case (state)
            half_wave_2:    if (controls_reg.pwm_factor == 0) //begin //if not used leaving state in fsm 
                                if (wave_cnt == controls_reg.frequency_div-1)
                                    if (perod_num_cnt == controls_reg.period_num)                           state = count;
                                    else                                                                    state = dead_time_2;
                                else                                                                        state = dead_time_2;
                            //end  //if not used leaving state in fsm 
                            else 
                                if (wave_cnt == controls_reg.frequency_div-controls_reg.pwm_factor-2)       state = ddr_edge_2;
                                else                                                                        state = half_wave_2;
            //
            ddr_edge_2:                                                                                     state = dead_time_2;
            dead_time_2:    if (wave_cnt == controls_reg.frequency_div-1)
                                if (perod_num_cnt == controls_reg.period_num)                               state = half_wave_2;
                                else                                                                        state = half_wave_2;
                            else                                                                            state = dead_time_2;
            count:                                                                                          state = half_wave_2;
            default:                                                                                        state = half_wave_2;
        endcase
        end // excess end is verilog error, but witout fsm is not recognized  
    end  // if always not used begin end fsm is not recognized  
end // if module not used begin end fsm is not recognized   
   
endmodule
**Code**
Code to reproduce the error.

**Please complete the following information:**
 - OS:Windows_NT x64 6.1.7601
 - VSCode version version 1.62.3

**Screenshots**

[image]

qarlosalberto, 2022-01-12:
I'm not Verilog user, but I think that it isn't possible to have a "case" with multiple statement and no "begin"-"end"

cbiker, 2022-01-13:
IEEE Std 1364-2005 9.5 Case statement
https://www.eg.bucknell.edu/~csci320/2016-fall/wp-content/uploads/2015/08/verilog-std-1364-2005.pdf
reg [15:0] rega;
reg [9:0] result;
case (rega)
16'd0: result = 10'b0111111111;
16'd1: result = 10'b1011111111;
16'd2: result = 10'b1101111111;
16'd3: result = 10'b1110111111;
16'd4: result = 10'b1111011111;
16'd5: result = 10'b1111101111;
16'd6: result = 10'b1111110111;
16'd7: result = 10'b1111111011;
16'd8: result = 10'b1111111101;
16'd9: result = 10'b1111111110;
default result = 'bx;
endcase