TerosTechnology/vscode-terosHDL #279: I install python  and other lib in Anconda ，but it does not work in windows

qgzln, 2021-11-29:
**Describe the bug**[image]A clear and concise description of what the bug is.
I install python  and other lib in Anconda ，but it does not work in windows
[image]

[image]
I alread excute pip install teroshdl
[image]

**To Reproduce**
**Code**
Code to reproduce the error.

**Please complete the following information:**
 - OS:windows10
 - VSCode version version 1.62.3

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

qarlosalberto, 2021-11-29:
Try to use: conda activate base & python3

You need to specific your python executable. TerosHDL runs: conda activate
base & python3 --version

qgzln, 2021-11-29:
> Try to use: conda activate base & python3 You need to specific your python executable. TerosHDL runs: conda activate base & python3 --version El lun., 29 nov. 2021 7:06, qgzln ***@***.***> escribió:
> […](#)

Thank you .but there is another issue:

[image]

qarlosalberto, 2022-01-12:
Can I close the issue?