TerosTechnology/vscode-terosHDL #280: Color problem of code block

zjulijinxin, 2021-12-15:
When I activate the plug-in, the "begin end" code block changes from multiple colors to only one color
[image]
[image]

qarlosalberto, 2021-12-16:
Can you share the complete code in text?

zjulijinxin, 2021-12-17:
Let me show a simple example, the following code:

`module test (
input   clk  ,
input   rst  ,
output  led  
);

always @(posedge clk) begin
  if (rst) begin
    led <= 0;
  end
  else begin
    led <= 1;
  end
end

endmodule`

When I activate the plug-in（TerosHDL），the "begin end" code block changes from multiple colors to only one color
[image]
[image]

**I think it's due to the conflict between the two plug-ins：TerosHDL and Verilog-HDL/SystemVerilog/Bluespec SystemVerilog**

qarlosalberto, 2021-12-17:
Yes, it seems a conflict with Verilog-HDL/SystemVerilog/Bluespec SystemVerilog. To check it, can you deactivate Verilog-HDL/SystemVerilog/Bluespec SystemVerilog ?

qarlosalberto, 2021-12-17:
And other thing, what extension do you use for the "begin" colors?

zjulijinxin, 2021-12-21:
After I deactivate this extension（Verilog-HDL/SystemVerilog/Bluespec SystemVerilog）, the code block has no different color：
[image]
I use the extension"Verilog HDL / SystemVerilog / bluespec SystemVerilog" to color code blocks.

Plus，I think the coloring function of the extension"Verilog HDL / SystemVerilog / bluespec SystemVerilog"   is not easy to use. It would be great if you could develop similar coloring functions.

qarlosalberto, 2022-01-12:
It will be fixed in the next version.

closed 2022-03-01 by zjulijinxin