TerosTechnology/vscode-terosHDL #281: Highlighting of module ports declaration for verilog codes is not correct

PascalZh, 2021-12-16:
**Describe the bug**
[image]
The last input is not correctly highlighted.

**Please complete the following information:**
Version: 1.63.1 (user setup)
Commit: fe719cd3e5825bf14e14182fddeb88ee8daf044f
Date: 2021-12-14T02:13:54.292Z
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Windows_NT x64 10.0.22000

qarlosalberto, 2021-12-16:
Can you share the code?

PascalZh, 2021-12-30:
```verilog
module foo (
  input a,
  input b,
  output c
);
  
endmodule
```
renders:
[image]

qarlosalberto, 2021-12-30:
got it! It will be fixed in the next version.

PascalZh, 2021-12-30:
It seems that github also doesn't render the syntax correctly. Maybe the verilog version is assumed to be verilog-2001, and verilog-2001 doesn't support the module declaration syntax above.

However, in my Quartus II (version 16.0.0) it works:
[image]

qarlosalberto, 2021-12-30:
I'm using the same syntax file as Github, so both fail. I have replaced the TerosHDL Verilog syntaxt by TerosHDL SV syntaxt, and it seems work fine.

closed 2022-03-25 by PascalZh