TerosTechnology/vscode-terosHDL #283: fsm isnt recognized - begin end bug

cbiker, 2021-12-21:
**Describe the bug**
>>>I found that this problem is caused by code irregularities
>>what is mean? code of example or code of teros?

now i see that not only always problem but its global problem with "begin end"
fsm recognized only if begin end at module at always and extra end at case(and code isnt right).

**To Reproduce**
module pattern_generator_stm ( bla bla );
begin // if module not used begin end fsm is not recognized
enum logic[3:0] {bla} state;
//! fsm_extract
always @(posedge clk )
begin // if always not used begin end fsm is not recognized
case (state)
half_wave_2: if (controls_reg.pwm_factor == 0) //begin //if not used leaving state in fsm
if (wave_cnt == controls_reg.frequency_div-1)
if (perod_num_cnt == controls_reg.period_num) state = count;
else state = dead_time_2;
else state = dead_time_2;
//end //if not used leaving state in fsm
else
if (wave_cnt == controls_reg.frequency_div-controls_reg.pwm_factor-2) state = ddr_edge_2;
else state = half_wave_2;
//
ddr_edge_2: state = dead_time_2;
dead_time_2: if (wave_cnt == controls_reg.frequency_div-1)
if (perod_num_cnt == controls_reg.period_num) state = half_wave_2;
else state = half_wave_2;
else state = dead_time_2;
count: state = half_wave_2;
default: state = half_wave_2;
endcase
end // excess end is verilog error, but witout fsm is not recognized
end // if always not used begin end fsm is not recognized
end // if module not used begin end fsm is not recognized

endmodule

**Please complete the following information:**
 - OS: Win7 sp1
 - VSCode version version 1.62.3

qarlosalberto, 2021-12-30:
Can you provide a correct code example?

qarlosalberto, 2021-12-30:
With no syntax errors.

closed 2023-06-26 by qarlosalberto