TerosTechnology/vscode-terosHDL #284: Neither icarus nor Modelsim can handle `include` directive

dzx-dzx, 2021-12-25:
**Describe the bug**
While attempting to compile my testbench...
with icarus:
```
Include file ./alu.v not found
error: Unable to find the root module "alu_tb" in the Verilog source.
     : Perhaps ``-s alu_tb'' is incorrect?
1 error(s) during elaboration.
```
with Modelsim:
```
Cannot find `include file "./alu.v" in directories:
#     C:/modeltech64_10.2c/ovm-2.1.2/../verilog_src/ovm-2.1.2/src, C:/modeltech64_10.2c/uvm-1.1d/../verilog_src/uvm-1.1d/src
# ** Error: C:/modeltech64_10.2c/win64/vlog failed.
# Executing ONERROR command at macro ./edalize_build_rtl.tcl line 3
make: *** [Makefile:42: work] Error 1
```
**Code**
Code to reproduce the error.
```verilog
`timescale 1ns/1ns
`include "./alu.v"
```
**Please complete the following information:**
 - OS: Windows
 - VSCode version 1.63.2

qarlosalberto, 2021-12-25:
Can you share your files?

dzx-dzx, 2021-12-25:
[alu.txt](https://github.com/TerosTechnology/vscode-terosHDL/files/7775920/alu.txt)
[alu_tb.txt](https://github.com/TerosTechnology/vscode-terosHDL/files/7775921/alu_tb.txt)

dzx-dzx, 2021-12-26:
Is there anything else I shall provide?

qarlosalberto, 2021-12-26:
Does it work if you provide the full path?

dzx-dzx, 2021-12-26:
> […](#)

That would be successful.

dzx-dzx, 2021-12-26:
Actually, does terosHDL simply not welcome the use of `include`? Few example projects have it.

qarlosalberto, 2021-12-26:
I need to check it.

dzx-dzx, 2021-12-27:
This might be helpful: Both files are placed inside a folder that is **not** the directory vscode opens.

qarlosalberto, 2021-12-30:
I have done a test with you files. You don't need to use "include" directive. You need to add both files to TerosHDL project manager:

[image]

The other solution is only add to the project manager the file "alu_tb.v" and to add the full path in the include directive:

[image]

dzx-dzx, 2021-12-30:
But...Using full path would render it impossible for publishing. Does that mean I simply can't use `include`?

qarlosalberto, 2021-12-30:
Yes, for now it isn't possible.

mtrx93, 2023-06-11:
Hello. 
Had same problem, there are at least two workarounds for icarus:

1. "-grelative-include" in "Additional arguments directly passed to linter."

2. "-I/absolute/path/to/project/" in same place, which allows to use includes relative to project directory (but suitable only for one project)

trexxet, 2024-02-18:
If you don't use `include`, then the linter goes mad about the unknown module type.
If you use `include`, then the Icarus just won't build it.

ecstrema, 2024-09-20:
Same here: cannot use the include directive at all. I want to integrate TerosHDL into an existing codebase, and so I can't remove all the include directives and add the files to the project. 

Overall, I wish Teros HDL worked with projects not created with it. I'd love to use it if it was incrementally adoptable/pluggable.

ooyawn, 2025-06-24:
FOR ICARUS: I use WSL: Ubuntu on VSCode. 
Go to your Settings on VSCode and look for Linting Include Paths. 
In my situation, the nature of code being outside of WSL where iVerilog is can confuse the Linter. 

Point out the desired directory to the Linter here (USING THE FILE SYSTEM IT IS STORED IN):
[image]

...and it will detect not only the file you want to include but all other files in said directory. 
No need to add the full path in the include directive on the code itself 👍🏻 

I apologize for my lack of proper technical verbiage. From what I understand, this crude diagram is how my setup works:
[image]

I hope the diagram properly shows how, without this step for a setup like mine, iVerilog can seemingly only work with one file at a time. 

This problem does not show up if both the Linter and desired directory are on the same system. 
It indeed does when the Linter is separated onto a subsystem (WSL).

P.S. I am aware of how cumbersome this setup is. 
My early professor had us work this way using C++ for security and safety. 
The downsides (like this very one) also include slower performance speed. 
As [this](https://learn.microsoft.com/en-us/windows/wsl/filesystems) guide highlights, I am working from a mounted drive. 
I would like to eventually transition to keeping my project files and programming tools in the same file system. 

If this does not help your situation I am sorry. @qarlosalberto 's tip about the full path helped me discover this solution for myself, thank you.