TerosTechnology/vscode-terosHDL #285: Not recognized state change when doing it inside a for loop

mfernandezt03, 2021-12-30:
**Describe the bug**
I'm creating a state machine that has to deal with four channels of input data. The machine changes it state according to the state of each chanel. Now, when I see the diagram, I don't find the changes between states that are instantiated inside the for loop created. 

**To Reproduce**
Create a state machine that handles different channels or inputs and changes it state according to that. 

**Code**
```
always@(posedge clk)begin
    case(state)
          caseA: state <= caseB;
          caseB: state <= case C;
          caseC: begin
                if(rx_data[0] == 8'hFF)begin
                      state <= caseD;
                      for( int k=1; k<4; k=k+1)begin
                           case(data[k])
                                   case1:  do_sth;
                                   case2:  do_sth;
                                   default: begin
                                          state <= caseZ;   // this is what does not appear in the diagram
                                   end
                           endcase
                      end
                end
          end
          more_cases:
          caseZ: state <= caseA;
    endcase
end
```

**Please complete the following information:**
 - OS: Ubuntu 20.04.3 LTS
 - VSCode version 1.63.2

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

[image]

qarlosalberto, 2021-12-30:
Could you provide a simple code to reproduce the problem? So I can debug it

qarlosalberto, 2021-12-30:
I see the problem. Currently the loops in the state machine aren't supported.

closed 2021-12-30 by mfernandezt03