TerosTechnology/vscode-terosHDL #286: Schematic viewer does not recognize GHDL on windows. (GHDL + Yosys)

HuskyNator, 2022-01-03:
**Describe the bug**
When I start the schematic viewer, the following error is thrown:
```
"C:\Program Files\yosys-win32-mxebin-0.9"\yosys.exe -p "ghdl --std=08 -fsynopsys e:\Code\Res-VHDL\Examples\Clock\clock_tb.vhd -e; proc; opt; write_json teroshdl_yosys_output.json; stat"
ERROR: No such command: ghdl (type 'help' for a command overview)

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2019  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9 (git sha1 1979e0b1, i686-w64-mingw32.static-g++ 5.5.0 -Os)

-- Running command `ghdl --std=08 -fsynopsys e:\Code\Res-VHDL\Examples\Clock\clock.vhd -e; proc; opt; write_json teroshdl_yosys_output.json; stat' --
2022-01-03 18:17:58: Configure (https://terostechnology.github.io/terosHDLdoc/netlist/configuration.html) Yosys or install YoWASP: pip install yowasp-yosys
ERROR: No such command: ghdl (type 'help' for a command overview)

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2019  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9 (git sha1 1979e0b1, i686-w64-mingw32.static-g++ 5.5.0 -Os)

-- Running command `ghdl --std=08 -fsynopsys e:\Code\Res-VHDL\Examples\Clock\clock_tb.vhd -e; proc; opt; write_json teroshdl_yosys_output.json; stat' --
2022-01-03 18:17:58: Configure (https://terostechnology.github.io/terosHDLdoc/netlist/configuration.html) Yosys or install YoWASP: pip install yowasp-yosys
```

Also note the strange `"C:\Program Files\yosys-win32-mxebin-0.9"\yosys.exe` path. _(I was unaware this was even valid)_
I was unable to escape the space using a `\`, and it is seemingly not done automatically, meaning the addition of `"`'s was necessary in the yosys binary path setting. I would suggest escaping this for windows.
This escape is however not the issue, changing it to a different directory retains the error.

I double checked & ghdl is defined in my path & works fine when called manually.

**To Reproduce**
1. Download the prebuilt yosys binary (http://bygone.clairexen.net/yosys/download.html) & set it in your path.
2. Have GHDL installed, on your path & in your theroshdl settings.
3. Create a smiple project & open one of the .vhd files (it can even be empty)
3. Start the schematic viewer. (using the Netlist viewer in the topright)
_For some reason, GHDL seems to be taken directly from the path, while the yosys settings has a preference for the manually defined path. Alternatively, GHDL setting _does_ properly escape spaces on windows._

**Code**
Does not matter, an empty .vhd file will even work.

**Please complete the following information:**
 - OS: Windows 10 Education - 21H1
 - VSCode version: 1.63.2
 - Yosys: prebuilt 0.9 binaries

HuskyNator, 2022-01-04:
Adding escapes to the GHDL setting will actually cause an error

qarlosalberto, 2022-01-12:
It seems that you Yosys installation doesn't include GHDL support. Try with:

https://github.com/ghdl/ghdl-yosys-plugin#pre-built-packages

closed 2023-06-26 by qarlosalberto