TerosTechnology/vscode-terosHDL #287: Error while running example project. run_edalize.py , line 28, in <module>     gui = sys.argv[4]

slava-li, 2022-01-04:
Hello 
On the Windows10pro Visual Studio Code v1.63.2 and TerosHDL 2.0.5 I'm getting error with "project_xsim" example (as well as with "project_vivado" example).
I tried to use Xsim and Vivado as a tool.
TerosHDL configuration check(for python etc.) is successful.

Could you please help with that? 

 

****************************************************************************************************************************************
---> Documentation: https://terostechnology.github.io/terosHDLdoc/project_manager/configuration.html
---> Project name: project_xsim
---> Top level: half_adder_process_tb    
---> Tool name: vivado     
---> Installation path: c:\Xilinx\Vivado\2016.2\bin\     
---> Open tool GUI: false
****************************************************************************************************************************************
2022-01-04 15:19:25: python "c:\Users\Corsair One\.vscode\extensions\teros-technology.teroshdl-2.0.5\resources\project_manager\run_edalize.py" "C:\Users\CORSAI~1\AppData\Local\Temp\edam.json" "vivado" "c:\Xilinx\Vivado\2016.2\bin\" "" "false" "vcdrom"
c:\Users\Corsair One\.vscode\extensions\teros-technology.teroshdl-2.0.5\resources\project_manager\run_edalize.py:6: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.spawn import find_executable

************************************************************************************************
---> Build directory: C:\Users\Corsair One\.teroshdl\build
Traceback (most recent call last):
  File "c:\Users\Corsair One\.vscode\extensions\teros-technology.teroshdl-2.0.5\resources\project_manager\run_edalize.py", line 28, in <module>
    gui = sys.argv[4]
IndexError: list index out of range
---> The test has finished with errors!
****************************************************************************************************************************************

slava-li, 2022-01-05:
In settings for Vivado it should be (for my PC)
c:\Xilinx\Vivado\2016.2\bin
instead of 
c:\Xilinx\Vivado\2016.2\bin\
Then there is no error.
However, i think this could be improved in TerosHDL.

I'm still learning TerosHDL  but looks like a great environment.
Thank you

qarlosalberto, 2022-01-12:
Hey @slava-li have you fixed your error? I don't understand:

```
c:\Xilinx\Vivado\2016.2\bin
instead of
c:\Xilinx\Vivado\2016.2\bin\
```

You have used the same path :S

SebastianHambura, 2022-04-20:
Hey, I can confirm this issue on Windows 11 and Vivado 2019.2 !

Using `C:\Xilinx\Vivado\2019.2\bin\` as installation path for XSim makes an error : 
```  
Traceback (most recent call last):
File "c:\Users\****\.vscode\extensions\teros-technology.teroshdl-2.0.6\resources\project_manager\run_edalize.py", line 28, in <module>
    gui = sys.argv[4]
IndexError: list index out of range
---> The test has finished with errors! 
```

However, it works once I removed the final `\` (so using `C:\Xilinx\Vivado\2019.2\bin` as installation path for XSim)

qarlosalberto, 2023-06-30:
fixed

closed 2023-06-30 by qarlosalberto