TerosTechnology/vscode-terosHDL #291: comment of 1st constant or signal in Architecture is ignored for documentation

thklumpp, 2022-01-14:
In this case:

```vhdl
...
ARCHITECTURE rtl OF AXI4_Error_Sync IS
  --! This comment will be ignored for documentation
  constant sync_ffs_c    : integer := 2;
...
```

qarlosalberto, 2022-01-14:
Your issues are very useful!! I will fix it, but I will take me a time... since I need to update the VHDL parser, and that is very hard.

thklumpp, 2022-01-14:
Ok.
And thanks a lot for this great extension!
Is there a workaround so far? A dummy comment or so?

qarlosalberto, 2022-01-14:
You need a VHDL statement. E.g:

[image]

qarlosalberto, 2022-01-14:
And you can select "only commented" if you want to delete de dummy signal/type/constant:

[image]

[image]

qarlosalberto, 2022-01-14:
In general you can see that you need a dummy statement (declaration, assign...) before the real first comment.

thklumpp, 2022-01-14:
Very helpful, thanks!

thklumpp, 2022-03-02:
Hi. I justed wanted to ask if there is a rough date for this fix. Or also, if this could be end of the year. Or already next month. 
It is only, that I can decide how to proceed to use TerosHDL or not - as the dummy assignments are not a nice option for me.
Anything helps. Thanks.

qarlosalberto, 2022-03-02:
It will be fixed this year, but not in the following months. Maybe at the
end.

qarlosalberto, 2023-06-26:
Fixed on v5.0.0

closed 2023-06-26 by qarlosalberto