TerosTechnology/vscode-terosHDL #292: Problem with VSG version > 3.3.0

elpollodiablo84, 2022-01-21:
**Describe the bug**
VSG formatter doesn't work when version is _strictly greater_ than 3.3.0

**To Reproduce**
Install a version of VSG _strictly greater_ than 3.30 (at this moment only 3.4.0 and 3.5.0), and try to use the formatter

**Versions**
 - Teros 2.0.6 or 2.0.5 
 - VSG > 3.30
 - VSCode 1.63.2

le-blue, 2022-01-23:
Just a (hopefully helpful) remark from someone who stumbled across this bug as well: 
In my case, VSG **worked** when installed on native python, i.e. outside of any virtual environment, but **did not work** when installed inside a virtualenv or anaconda env.

# 

**Versions**
- Windows 10
- VSCode 1.63.2
- Teros 2.0.6
- Python 3.10.2
- VSG 3.5.0

elpollodiablo84, 2022-01-24:
Sadly in my case VSG is installed on native python and it doesn't work on version > 3.3.0. for the sake of completeness, my python version is 3.8.3.

kele14x, 2022-04-08:
Seems I got same issue. TerosHDL output is:

```plain
****************************************************************************************************************************************
2022-04-08 10:47:54: Formatting file with vsg: 
****************************************************************************************************************************************
```

and nothing happens on editing file.

elpollodiablo84, 2022-09-08:
I've got some good news about this issue: apparently the change that broke the integration is in the rules configuration file, in facts using vsg and teroshdl without a custom yaml configuration file works as intended in versions > 3.3.
Basically there's nothing wrong with teroshdl, so I think the issue should be closed.

closed 2022-09-08 by elpollodiablo84