TerosTechnology/vscode-terosHDL #293: Verible ruleset not interpreted correctly

suzizecat, 2022-01-25:
**Describe the bug**
When specifying rules changes to VeribleLint parameters, Teros will remove comma separating rules while those shall be kept e.g.
`-no-trailing-spaces,-parameter-name-style` provides 
```text
verible-verilog-lint --rules -no-trailing-spaces -parameter-name-style
```
instead of
```text
verible-verilog-lint --rules -no-trailing-spaces,-parameter-name-style
```

**To Reproduce**
Use verible linter and specify rules in the "what rules to use"

**Please complete the following information:**
 - OS: Ubuntu 20.04
 - VSCode version :
 ```text
Version: 1.63.2
Commit: 899d46d82c4c95423fb7e10e68eba52050e30ba3
Date: 2021-12-15T09:39:46.686Z
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Linux x64 5.4.0-96-generic snap
```

proto17, 2023-04-21:
I got this working by changing the `get_veriblelint_options` function in `resources/project_manager/config2.html` from

```
            function get_veriblelint_options() {
                let options = {
                    'veriblelint': {
                        'installation_path': document.getElementById("veriblelint_installation_path").value,
                        'ruleset': document.getElementById("veriblelint_ruleset").value,
                        'verible_lint_args': document.getElementById("veriblelint_verible_lint_args").value.split(','),
                        'rules': document.getElementById("veriblelint_rules").value.split(','),
                    }
                };
                return options;
            }
```

to

```
            function get_veriblelint_options() {
                let options = {
                    'veriblelint': {
                        'installation_path': document.getElementById("veriblelint_installation_path").value,
                        'ruleset': document.getElementById("veriblelint_ruleset").value,
                        'verible_lint_args': document.getElementById("veriblelint_verible_lint_args").value.split(','),
                        'rules': document.getElementById("veriblelint_rules").value,
                    }
                };
                return options;
            }
```

Note that the only change was the `rules` line no longer having a `split(',')` call

This can be tried out on an existing install by editing (assuming Linux) `~/.vscode/extensions/teros-technology.teroshdl-*/resources/project_manager/config.html` and making the same change as above.  Restart VSCode and you should see the correct things happening.

qarlosalberto, 2023-06-26:
fixed on v5.0.0

closed 2023-06-26 by qarlosalberto