TerosTechnology/vscode-terosHDL #296: Schematic viewer build failed when the Verilog file use `include files`

HaitianJiang, 2022-01-31:
Here I have two files:

**IF_stage.v**

```verilog
`include "mycpu.h"

module if_stage(
    input                          clk            ,
    input                          reset          ,
    //allwoin
    input                          ds_allowin     ,
    //brbus
    input  [`BR_BUS_WD       -1:0] br_bus         ,
    //to ds
    output                         fs_to_ds_valid ,
    output [`FS_TO_DS_BUS_WD -1:0] fs_to_ds_bus   ,
    // inst sram interface
    output        inst_sram_en   ,
    output [ 3:0] inst_sram_wen  ,
    output [31:0] inst_sram_addr ,
    output [31:0] inst_sram_wdata,
    input  [31:0] inst_sram_rdata
);

...

endmodule
```

I'm sure that this file have no bugs.

The include file is:
**mycpu.h**
```cpp
`ifndef MYCPU_H
    `define MYCPU_H

    `define BR_BUS_WD       32
    `define FS_TO_DS_BUS_WD 64
    `define DS_TO_ES_BUS_WD 136
    `define ES_TO_MS_BUS_WD 71
    `define MS_TO_WS_BUS_WD 70
    `define WS_TO_RF_BUS_WD 38
`endif
```

OS: Windows 10
Schematic viewer: YoWASP

When I run the viewer,it will promt me **ERROR: Can't open include file `mycpu.h'!**.

I have add two files in my TerosHDL project,and I test use full path, it failed too.

What happend?Does the TerosHDL not support include command?

beanspower, 2023-07-14:
@qarlosalberto 
Dear expert,
The same warning also show  when the linter in enabled
[image]
The linter configuration is shown as below
[image]

Besides, if use ` include` all the other style checker or linter messages will not display for below RTL anymore.
[image]
[image]

beanspower, 2023-07-14:
I use one workaround for this issue.
Take the reference from :
1.https://stackoverflow.com/questions/47449489/how-to-include-files-in-icarus-verilog
2.https://github.com/mshr-h/vscode-verilog-hdl-support/issues/51

I add -i and -y arguments in configurations ---- linter settings ----- Linter settings: icarus linter --- Arguments
`-I C:\01_Modelsim\inc -I C:\ipcore_dir `
[image]

qarlosalberto, 2023-07-14:
@beanspower can you provide a complete example with code?

beanspower, 2023-07-14:
@qarlosalberto Hello expert，
I try to provide one example code for it.
Thanks for your kind help
[image]
[Example.zip](https://github.com/TerosTechnology/vscode-terosHDL/files/12051085/Example.zip)