TerosTechnology/vscode-terosHDL #300: Include module in top.v and Hierarchy view

mlogelect, 2022-02-12:
Hi,
there are two things i can't resolve. First of all i can't include more submodule inside a top module, in this photo you can see the barrett_round (top) and the two module that he needs (adder and multiplier), but when i try to add them the IDE shows that red underline. Am i missing something? 

I don't know if this is correlated to the second problem: in hieararchy view it says "Configure your pyhton 3...." but i already did the "pip install teroshdl, python 3 is alredy installed 

[image]

qarlosalberto, 2022-02-13:
For Verilog/SV the linter only checks the errors in one file, so you will see errors if you include one module in other. It will fix in the future.

qarlosalberto, 2022-02-13:
The Schematic Viewer and the Linter are independent features. You will see an error in the Linter, but you won't have problems with the Schematic Viewer.

mlogelect, 2022-02-13:
Ok, thank you for your fast response, it works! I close the issue

closed 2022-02-13 by mlogelect