TerosTechnology/vscode-terosHDL #302: Additional iverilog options are not passed for linting

o-alexandre-felipe, 2022-02-16:
Take this example file
``` sv
module a;
logic p[1];
endmodule
```

Teros HDL is clearly saying that a command line option would make the declaration of `p` acceptable
[image]

I tried to set using the configuration menu
[image]

Development container:
Icarus Verilog version 11.0 (stable) (84b4ebe)
Linux 5b6ebb0fadde 4.19.104-microsoft-standard #1 SMP Wed Feb 19 06:37:35 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

Host:
[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

qarlosalberto, 2022-02-16:
For the linter you need to set the parameters in the last box: "Additional arguments passed to the linter..."

qarlosalberto, 2022-02-17:
@o-alexandre-felipe does it resolve your problem?

o-alexandre-felipe, 2022-04-12:
Hi sorry for the late response, that solves this issue.

closed 2022-04-12 by o-alexandre-felipe