TerosTechnology/vscode-terosHDL #303: [feature] Specify the build directory in the current workspace.

hwhsu1231, 2022-02-22:
Can TerosHDL provide the settings to let us specify the `build` directory?
By default, it will be generated at `C:\Users\<username>\.teroshdl\build`.
But I want to generate at `${workspaceFolder}/build`.

qarlosalberto, 2022-02-22:
It's a good idea.

hakan-demirli, 2022-08-20:
I think on top of allowing users to change the build location TerosHDL should utilize the current working directory of the VSCode as a default directory for builds. Or use a build subfolder relative to the project.yml as a default for that specific project.

Also, if you make the project.yml path an environment or internal variable, people can pass it to the other linting tools (for example icarus -y $project_path). This then can be used to solve these: https://github.com/TerosTechnology/vscode-terosHDL/issues/325 types of issues.

This can also help the waveform viewer. Currently, vcd files are generated at `.teroshdl/build` directory. It's not possible to directly see the output files in VSCode if the directory is not open. You have to change your working directory. Hence, if the build folder is in a sub-directory, the problem is solved. 

Not only that, currently,  everything in the build directory is overwritten when a new build starts. So, to compare the waveforms of different runs, you have to manually backup the output files. This would be solved too.

I also occasionally run automated scripts to parse and compare the custom outputs of different testbenches. As you can understand, manually backing things up before the next build is the only option right now. That's why I still haven't completely switched to TerosHDL.

qarlosalberto, 2022-08-21:
Thanks for the long description. I will implement

krushna-katkar, 2025-11-30:
hey i need to change the directory because there is an error when i try to run the testbench.And the error is my name in <User>, i have a space break in my name so teroshdl gets confused. see the file to understand the error

[image]