TerosTechnology/vscode-terosHDL #309: Modelsim: project file order breaks project if not sorted like hierarchy using

maor1993, 2022-04-08:
**Describe the bug**
modelsim will fail to run package imports if the order is not as imported, packages have to be first in the list 

**To Reproduce**
* make a  package file 
* make an sv file which imports the file
* place the sv file **before** the package file
* try to run modelsim on sv file

**Please complete the following information:**
 - OS: windows 10
 - VSCode version version 1.66.1

**Screenshots**
trying to run with mips consts in wrong order (imported by if_stage_top.sv)
[image]
```
# do edalize_main.tcl
# ** Error: c:/Users/maor1/gitlab/verimips/RTL/MIPS/IF/if_stage_top.sv(32): (vlog-13006) Could not find the package (mips_consts).  Design read will continue, but expect a cascade of errors after this failure.  Furthermore if you experience a vopt-7 error immediately before this error then please check the package names or the library search paths on the command line.
# ** Error: c:/Users/maor1/gitlab/verimips/RTL/MIPS/IF/if_stage_top.sv(111): (vlog-2164) Class or package 'mips_consts' not found.
# ** Error: c:/Users/maor1/gitlab/verimips/RTL/MIPS/IF/if_stage_top.sv(111): (vlog-2730) Undefined variable: 'mips_consts'.
# ** Error: (vlog-13069) c:/Users/maor1/gitlab/verimips/RTL/MIPS/IF/if_stage_top.sv(111): near "::": syntax error, unexpected ::, expecting ';' or ','.
# ** Error: C:/modeltech64_10.7/win64/vlog failed.
# Executing ONERROR command at macro ./edalize_build_rtl.tcl line 5
make: *** [work] Error 1
```

fixing order by removing the files:
[image]
```
# ** Note: $finish    : c:/Users/maor1/gitlab/verimips/Testing/RTL/MIPS/IF/if_stage_top_tb.sv(38)
#    Time: 110 ns  Iteration: 0  Instance: /if_stage_top_tb
# End time: 17:10:54 on Apr 08,2022, Elapsed time: 0:00:02
# Errors: 0, Warnings: 4
```

maor1993, 2022-04-10:
Also occurs with icarus.

qarlosalberto, 2022-04-10:
Yes, it happens in all the tools. I wi try to fix it I'm the next release.

maor1993, 2022-04-11:
What's the plan? Maybe I can help?

qarlosalberto, 2022-04-11:
I will fix it in the following days and I will send you a beta version to check it.

qarlosalberto, 2022-04-11:
Could you send me an email to carlosruiznarnajo@gmail.com ?