TerosTechnology/vscode-terosHDL #313: Some thoughts on improvements

cng18, 2022-04-11:
Hello Developers,
First of all my sincere thanks for putting together this extension for vscode. Over the past few days I played around teroshdl-vscode. I would say that its a great step in bringing some order to the HW design flows. I experimented with the document generator and found it to be a great motivator for adding meaningful comments to the RTL code. I tried running Lint and synthesis using various tools and got it working after playing around with the settings a bit. During this trials I noted my findings. I feel that the whole thing can be made more intuitive. Let me explain what I mean:

I would breakdown a HDL design into following steps. A design has to be certified for all the Steps shown.

1. RTL Code
2. RTL Style Check
3. RTL Lint Check
4. RTL CDC/RDC checks
5. Simulation/Verification
6. Synthesis
7. DFT checks
8. IPXACT of Design (ports/interfaces/registers)

For each of the above steps we can choose one of the many available tools. And for the chosen tool we should be able to provide options.

So we have a Design Flow Step followed by the Tool to be used for that Step  followed by Options to be passed to the tool for running that specific step.

Below I try to illustrate using example:

Example Steps:
Step=Lint Check  ---> Tool=Verilator   ---> Verilator Options for Lint Check  ---> Run Lint using Verilator
Step=Lint Check  ---> Tool=Vivado xvlog ----> Vivado xvlog options for Lint Check ----> Run Lint using Vivado Xvlog
Step=Verification ---> Framework=Cocotb ---> Tool = Verilator ----> Verilator options for Simulation 

1. Tool Configuration Window can be renamed as Design Flow Configuration Window.
2. In the Design Flow Configuration Window you choose the Step-Tool-Options and add it to the Runs List
3. In the Runs List now we have multiple Steps Listed with a Run Button against each step and also a Status Bar.
4. A step can be repeated in the Runs List. For example the Lint Check can be done with various tools.
5. The Runs List also acts like a dash board for the end user.
6. The user specified build directory will have a structure like
- build --> lint step --> tool1 ---> reports
- build --> lint step --> tool2 ---> reports

Just wanted to share my thoughts.  :-)

qarlosalberto, 2022-04-11:
That's cool! Could you send me an email to carlosruiznaranjo@gmail.com

qarlosalberto, 2022-04-12:
I would like to discuss about the workflow in TerosHDL.