TerosTechnology/vscode-terosHDL #317: Error include makefiles in cocotb framework

Warl0ck73, 2022-05-08:
When I include a makefile, using a relative path or the computed variable then run list is empty. If I use an absolute path to include the makefile, all is well and the run list contains the tests.
I think: it doesn't should parse files include by same method.

**To Reproduce**
PWD := $(shell pwd)
include $(PWD)/Makefile1
include ./Makefile2

**Please complete the following information:**
 - OS: Windows 10 x64
 - VSCode version version 1.67.0

**Screenshots**

[image]

[image]

qarlosalberto, 2022-05-08:
I will take a look.

Warl0ck73, 2022-05-09:
Sorry, it's my mistake.. It's because of the incompatibility path Windows and UNIX