TerosTechnology/vscode-terosHDL #320: Don

WTFive, 2022-05-17:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Code to reproduce the error.

**Code**
Code to reproduce the error.

**Please complete the following information:**
 - OS: [e.g.Ubuntu 18.04]
 - VSCode version version [e.g. 22]

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

qarlosalberto, 2022-06-28:
Does it work fine the "outline"?

qarlosalberto, 2022-06-28:
Ok, I see the problem. I will fix it.

ufktmr, 2022-11-18:
im having the same problem in win10 vscode version 1.73.1 and terosHDL version v2.0.7

qarlosalberto, 2022-11-18:
@ufktmr could you send me an email? carlosruiznaranjo@gmail.com

alpetrovsky774, 2024-04-01:
This is still happening in version 5.0 and the pre-release version 6.0. The impact is that autocomplete, go-to definition, hover, and evaluate features do not work for most elements in the Verilog/SV files as those features depend on identification of element names in the file. It does however seem to work fine with VHDL. 

For example:
input wire [7:0] input_a; // parsed incorrectly, the keyword "wire" is identified as the element name
output reg [7:0] output_a; // parsed incorrectly, the keyword "reg" is identified as the element name
...
reg [1:0] unsigned_reg; // parsed correctly and name "unsigned_reg" is properly identified
reg signed [1:0] signed_reg; // parsed incorrectly, the keyword "signed" is identified as the element name

Based on the above examples it seems that the word following reg/wire or input/output keywords is used as the element name regardless of where the given name is actually located within the line. This prevents autocomplete, go-to definition, hover, and evaluate features from working on most input/output definitions and signed reg/wire definitions.