TerosTechnology/vscode-terosHDL #321: verible-verilog-format not works

zfling, 2022-05-22:
**Describe the bug**
verible-verilog-format not works

**Please complete the following information:**
 - OS: Ubuntu 20.04
 - VSCode version version 1.67.2

**Screenshots**
[image]
not work
[image]
[image]
work
[image]
how to set the current file?

qarlosalberto, 2022-05-22:
I will take a look.

zfling, 2022-05-22:
[image]
Nothing to do!

michael-etzkorn, 2022-07-19:
You're supposed to separate the arguments by commas.

closed 2023-06-26 by qarlosalberto