TerosTechnology/vscode-terosHDL #324: Can't find definitions for localparam, but it can for parameter

MasterJubei, 2022-06-28:
Hello, great project

One small bug I found is that it cannot find the definitions of localparam variables

[image]

But if I set it to parameter, it will work

[image]

qarlosalberto, 2023-06-29:
could you provide a code example?

MasterJubei, 2023-07-01:
> 

Hey not sure, but it looks like it has been fixed(or I can't reproduce it). Both localparam and parameter both show the definition now.

`module localparam_test();

    localparam NUM_BITS = 3;
    // parameter NUM_BITS = 3;
    
    integer i = 0;
    initial begin
        for(i = 0; i < NUM_BITS; i = i+ 1) begin
            $display("i = %d", i);
        end
    end
endmodule`

closed 2023-07-01 by MasterJubei