TerosTechnology/vscode-terosHDL #331: 'show netlist' cannot find the top module

GrootOfSDUST, 2022-07-21:
when i use the show netlist button to use yosys, I found the plugin always show the sub module

So I modified the source code of the plugin in the file :` teros-technology.teroshdl-2.0.7/out/lib/netlist_viewer/netlist_viewer.js`
to identify the top moule

- before

```var script_code;
const script_code = `${cmd_files}; hierarchy -top top; proc; opt; write_json ${output_path_filename}; stat`;
```

- after

```var script_code;
if (cmd_files === "read_verilog -sv top.v")
{
    script_code = `${cmd_files}; hierarchy -top top; proc; opt; write_json ${output_path_filename}; stat`;
}
else
{
    script_code = `${cmd_files}; proc; opt; write_json ${output_path_filename}; stat`;
}
```
but to use this function, we must name the top module by top.

[image]

qarlosalberto, 2022-07-21:
Could you share a reproducible example? (In code)

GrootOfSDUST, 2022-07-21:
This is my structure
[image]
the plugin will always set the id.v as top module before i modify the code

qarlosalberto, 2022-07-21:
Have 1 file or more?
What button are you using?
Are you using the project manager?

GrootOfSDUST, 2022-07-21:
the button i used is 'nestlist viewer'
[image]

and i used the project manager
[image]

qarlosalberto, 2022-07-21:
In the project manager, rigth? There are 2 buttons:

https://terostechnology.github.io/terosHDLdoc/netlist/start.html

https://terostechnology.github.io/terosHDLdoc/project_manager/hierarchy.html

GrootOfSDUST, 2022-07-21:
the plugin's yosys dosen't have the capability to find the top module, so I edited the source code as I showed before.

GrootOfSDUST, 2022-07-21:
> In the project manager, rigth? There are 2 buttons:
> 
> https://terostechnology.github.io/terosHDLdoc/netlist/start.html
> 
> https://terostechnology.github.io/terosHDLdoc/project_manager/hierarchy.html

i used the first one, because the second will show nothing to me.
[image]

and, if i click the first 'netlist viewer', the schemetic viewer won't show netlist before I switch the schmetic viewer to anthor and switch back
I will show you tomorrow after I make this bug to a GIF

qarlosalberto, 2022-07-24:
Please, share a complete example in **code**, not an image. So I can reproduce it.

qarlosalberto, 2022-10-11:
Okk. I could take a look the next week. It would be really useful if you
can share all your project. Or a minimum example. In other case it's
difficult for me reproduce your error.

qarlosalberto, 2023-07-03:
fixed

closed 2023-07-03 by qarlosalberto