TerosTechnology/vscode-terosHDL #339: Not detecting Make

vani2620, 2022-09-21:
I installed Make already, but when I check the configuration, the plugin tells me it's not installed. I have no idea why. If it's helpful, [here](https://pastebin.com/C4em4RzF) is the terminal output. 

The path to Make is C:\Program Files (x86)\GnuWin32 if that helps any.

vani2620, 2022-09-21:
Update: Tried adding Make bin to path, no dice.

qarlosalberto, 2022-09-21:
Does it work if you type in the windows terminal "make --version"?

vani2620, 2022-09-21:
Kind of?

What I get is [this](https://pastebin.com/Py2dxNei).

qarlosalberto, 2022-09-21:
What do you have storage characters and a "make error"? Maybe that's the
problem

vani2620, 2022-09-21:
I genuinely don't know what's causing that.

qarlosalberto, 2022-09-21:
TerosHDL calls to "make --version" and checks the return value. If it's
failing the renturn value won't be 0 an it will fail.

qarlosalberto, 2022-09-24:
I'm sorry, I'm not Windows user.

LippiD, 2022-12-23:
@vani2620 Did you achieve in solving your problem? I'0ve got something similar. I installed make, but Theroshdl doesn't see it.
I have also added its path to the PATH environment variable but without success.