TerosTechnology/vscode-terosHDL #340: Nesting case statements avoids FSM to be detected

Tobi-Alonso, 2022-09-26:
**Description of the bug**
The state machine viewer does not detect an FSM if nested case statements are employed.

**Code To Reproduce**
While an FSM is detected for arch_0, it's not for arch_1:
``` vhdl
library ieee;
  use ieee.std_logic_1164.all;

ENTITY state_machine IS
  PORT(
    clk      : in   std_logic;
    input    : in   std_logic_vector(1 downto 0)
  );
end state_machine;

architecture arch_0 of state_machine is
  type state_type is (STATE_0, STATE_1, STATE_2);
  signal state, next_state   : state_type;
begin

  process (clk)
  begin
    if rising_edge(clk) then
      state <= next_state;
    end if;
  end process;

  next_state_proc: process( state, input)
  begin
    case( state ) is
      when STATE_0 =>
        state <= STATE_1;
      when STATE_1 =>	
        state <= STATE_2;
      when STATE_2 =>
        if input = "00" then
          state <= STATE_1;
        elsif input = "01" then
          state <= STATE_2;
        else
          state <= STATE_0;
        end if;

    end case;      
  end process;
end arch_0 ;

architecture arch_1 of state_machine is
  type state_type is (STATE_0, STATE_1, STATE_2);
  signal state, next_state   : state_type;
begin

  process (clk)
  begin
    if rising_edge(clk) then
      state <= next_state;
    end if;
  end process;

  next_state_proc: process( state, input)
  begin
    case( state ) is
      when STATE_0 =>
        state <= STATE_1;
      when STATE_1 =>	
        state <= STATE_2;
      when STATE_2 =>
        case( input ) is
          when "00" =>
            state <= STATE_1;
          when "01" =>
            state <= STATE_2;
          when others =>
            state <= STATE_0;
        end case;

    end case;      
  end process;
end arch_1;

```

(Detected with tested code)

**Setup:**
 - OS: Windows 11
 - VSCode version version: 1.71.2

qarlosalberto, 2023-06-26:
TerosHDL doesn't support `case` inside `case`

EdwinEstep, 2023-07-13:
Are there plans to add this feature in the future? I know it used to allow case statements that didn't change the state, but those don't seem to be accepted at all anymore. 

As an example, I modified your `fsm_6.vhd` example state machine. It can handle when there are no nested case statements:
[image]

But it fails here, even though the case statement can't affect state changes:
[image]

qarlosalberto, 2023-07-13:
Yes, but not in the near future.