TerosTechnology/vscode-terosHDL #341: [QUESTION]: Automatic update files in project?

sibeov, 2022-09-30:
**Describe the bug**
This is more of a question. Is it possible to access som API or similar to be able to automatically update or add files to the project.
ATM i currently have a script generating a .csv file which I have to add everytim I add new files.

Even beeing able to use a command via the command palett would be nice!

**To Reproduce**
NA

**Code**
NA

**Please complete the following information:**
 - OS: Win10
 - VSCode version 1.71.2

**Screenshots**
NA

**Additional context**
NA

qarlosalberto, 2022-09-30:
Currently it's not possible. But your suggestion is very interesting I will take it in count

qarlosalberto, 2023-06-26:
Done in v5.0.0

closed 2023-06-26 by qarlosalberto