TerosTechnology/vscode-terosHDL #342: VSCode fails to find yowasp-yosys

afbcom, 2022-10-04:
**Error**
The error occurs when trying to use schematic viewer.
State machine viewer, module docmentation preview, generate template all work as expected.

VSCode with terosHDL extension fails to find the yowasp-yosys executable.  The output is:
```
yowasp-yosys -p "read_verilog -sv mk_mpipe.v; proc; opt; write_json teroshdl_yosys_output.json; stat"
/bin/sh: yowasp-yosys: command not found
2022-10-04 08:32:32: Configure (https://terostechnology.github.io/terosHDLdoc/netlist/configuration.html) Yosys or install YoWASP: pip install yowasp-yosys
```

**Steps taken**
Install VSCode, add terosHDL extension (also remote-SSH, SystemVerilog Language Support). terosHDL extension is installed _inside_ remote-SSH extension.
To install yowasp-yosys:
Install python 3.10, run pip install yowasp-yosys
Add to path
Successfully installed, can call from command line.

**Please complete the following information:**
 - OS: Windows 11 (also replicated on Windows 10
 - VSCode version: 1.71.2
 - Date: 2022-09-14T21:03:37.738Z
 - Electron: 19.0.12
 - Chromium: 102.0.5005.167
 - Node.js: 16.14.2
 - V8: 10.2.154.15-electron.0
 - OS: Windows_NT x64 10.0.22000
 - Python 3.10.7
 - yowasp-yosys: 0.21.dev414

**Screenshots**
N/A

**Additional context**
Also installed teroshdl via pip:
I have tried setting the path to yosys (with both forward slash and double backslash), similar but not same error:
```
C:/Users/afbcom/Downloads/oss-cad-suite/yosys -p "read_verilog -sv ncel2c500.v; proc; opt; write_json teroshdl_yosys_output.json; stat"
/bin/sh: C:/Users/afbcom/Downloads/oss-cad-suite/yosys: No such file or directory
2022-10-04 08:43:12: Configure (https://terostechnology.github.io/terosHDLdoc/netlist/configuration.html) Yosys or install YoWASP: pip install yowasp-yosys
```

The error occurs when trying to use schematic viewer.
State machine viewer, module docmentation preview, generate template all work as expected.

qarlosalberto, 2022-10-14:
Open a CMD console an type: ```yowasp-yosys --help```
does it work?

afbcom, 2022-10-14:
It does for me.

qarlosalberto, 2022-10-14:
Try both, please

afbcom, 2022-10-14:
They both work.

qarlosalberto, 2022-10-14:
You are using TerosHDL in a remote machine, right? Is it installed in the remote machine?

afbcom, 2022-10-14:
Sorry my original report wasn't clear.  It's all local, the ssh functionality just let's you browse a remote filesystem.  I also verified the same behaviour without the remoteSSH plugin on both Windows 10 and 11.

tthrx, 2022-10-20:
> Open a CMD console an type: `yowasp-yosys --help` does it work?

it doesn't work both in CMD and in PowerShell

tthrx, 2022-10-28:
found the reason, yowasp-yosys is not in system path.
for my case the yowasp-yosys is installed using pip install teroshdl, and the default path is 
"C:\Users\xxx\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts"
add this to system path, the schematic viewer works nomally.
it's not a bug

qarlosalberto, 2022-10-28:
cool!

closed 2023-06-30 by qarlosalberto

Huyhq29-AI, 2025-04-12:
WINDOW: I used teros with schematic viewer on VScode and these error pops up. Please help
2025-04-12 10:13:08.210 [info] Executing: C:\WINDOWS\system32\cmd.exe  C:\WINDOWS\system32\cmd.exe /d /s /c " yowasp-yosys -p "read_verilog -sv "h:/HK242/VERON/TEI0003_CYC1000/RTL sample/uart_i2c/uart_i2c.sv"; hierarchy -top uart_i2c; proc; write_json .teroshdl_BSNGX; stat""
2025-04-12 10:13:08.657 [info] 

2025-04-12 10:13:08.658 [info] ERROR: Can't guess frontend for input file `sample/uart_i2c/uart_i2c.sv; hierarchy -top uart_i2c; proc; write_json .teroshdl_BSNGX; stat' (missing -f option)!

2025-04-12 10:13:08.697 [error] Yosys failed.