TerosTechnology/vscode-terosHDL #346: Adding cocotb Makefile to project does nothing

threewholefish, 2022-10-25:
**Describe the bug**
The [documentation](https://terostechnology.github.io/terosHDLdoc/project_manager/start.html) says that for cocotb project, you only need to add the Makefile to the TerosHDL project. I presumed this would read dependencies, etc., but nothing seems to happen.  

It did not work with my personal project, so I tried the [cocotb matrix multiplier example](https://github.com/cocotb/cocotb/tree/master/examples/matrix_multiplier/tests), which did not work either.

The configuration check claims that cocotb is installed:
```
****************************************************************************************************************************************
---> Python3 path: /home/user/.conda/envs/teroshdl/bin/python
---> Python3 packages directory: /home/user/.conda/envs/teroshdl/lib/python310.zip,/home/user/.conda/envs/teroshdl/lib/python3.10,/home/user/.conda/envs/teroshdl/lib/python3.10/lib-dynload,/home/user/.conda/envs/teroshdl/lib/python3.10/site-packages
---> VUnit is installed.
---> Cocotb is installed.
---> yowasp-yosys is installed.
---> Edalize is installed.
---> Make is installed.
****************************************************************************************************************************************
```

**To Reproduce**
* Clone [cocotb](https://github.com/cocotb/cocotb/)
* Create new TerosHDL project
* Open [matrix multiplier example Makefile](https://github.com/cocotb/cocotb/blob/master/examples/matrix_multiplier/tests/Makefile)
* 
**Please complete the following information:**
 - OS: Ubuntu 20.04
 - VSCode version 1.72.2

qarlosalberto, 2022-10-26:
Have you selected as tool cocotb??

https://terostechnology.github.io/terosHDLdoc/configuration/tools.html#tools

threewholefish, 2022-10-26:
Yes, and I have set the cocotb binary directory to be the same directory as
Python3 binary.