TerosTechnology/vscode-terosHDL #349: Feature request - TerosHDL settings per project

dramoz, 2022-11-03:
Provide independent TerosHDL settings per project.
- Currently, tools are set globally, but sometimes different projects require different tools, or it would be required to test on different tools the functionality of testcases in different environments.
- Settings ala VS code style could be a nice approach, where there are global/user settings and then workspace/project settings
- Also provide a way to backup TerosHDL settings (git) [#312](https://github.com/TerosTechnology/vscode-terosHDL/issues/312#issue-1198977152)
  - It could be done by adding it to VS code sync settings or by using the proposed setting per project

PeterWang-dev, 2024-03-06:
Any progress on this issue?

qarlosalberto, 2024-03-06:
Yes, in the pre-release version you can use the settings per project.