TerosTechnology/vscode-terosHDL #351: Schematic viewer does not work in WSL2

small-han, 2022-11-21:
When I open Schematic viewer, TerosHDL tells me following.It seems viewer works well ,but I cant see any graph.

yowasp-yosys` -p "read_verilog -sv test.sv; proc; opt; write_json teroshdl_yosys_output.json; stat"

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.23 (git sha1 UNKNOWN, ccache clang 11.0.0-2~ubuntu20.04.1 -Os -flto -flto)

-- Running command `read_verilog -sv test.sv; proc; opt; write_json teroshdl_yosys_output.json; stat' --

1. Executing Verilog-2005 frontend: test.sv
Parsing SystemVerilog input from `test.sv' to AST representation.
Generating RTLIL representation for module `\test'.
Successfully finished Verilog frontend.

2. Executing PROC pass (convert processes to netlists).

2.1. Executing PROC_CLEAN pass (remove empty switches from decision trees).
Cleaned up 0 empty switches.

2.2. Executing PROC_RMDEAD pass (remove dead branches from decision trees).
Removed a total of 0 dead cases.

2.3. Executing PROC_PRUNE pass (remove redundant assignments in processes).
Removed 0 redundant assignments.
Promoted 0 assignments to connections.

2.4. Executing PROC_INIT pass (extract init attributes).

2.5. Executing PROC_ARST pass (detect async resets in processes).

2.6. Executing PROC_ROM pass (convert switches to ROMs).
Converted 0 switches.
<suppressed ~1 debug messages>

2.7. Executing PROC_MUX pass (convert decision trees to multiplexers).
Creating decoders for process `\test.$proc$test.sv:6$1'.
     1/1: $0\count[0:0]

2.8. Executing PROC_DLATCH pass (convert process syncs to latches).

2.9. Executing PROC_DFF pass (convert process syncs to FFs).
Creating register for signal `\test.\count' using process `\test.$proc$test.sv:6$1'.
  created $dff cell `$procdff$5' with positive edge clock.

2.10. Executing PROC_MEMWR pass (convert process memory writes to cells).

2.11. Executing PROC_CLEAN pass (remove empty switches from decision trees).
Found and cleaned up 1 empty switch in `\test.$proc$test.sv:6$1'.
Removing empty process `test.$proc$test.sv:6$1'.
Cleaned up 1 empty switch.

2.12. Executing OPT_EXPR pass (perform const folding).
Optimizing module test.

3. Executing OPT pass (performing simple optimizations).

3.1. Executing OPT_EXPR pass (perform const folding).
Optimizing module test.

3.2. Executing OPT_MERGE pass (detect identical cells).
Finding identical cells in module `\test'.
Removed a total of 0 cells.

3.3. Executing OPT_MUXTREE pass (detect dead branches in mux trees).
Running muxtree optimizer on module \test..
  Creating internal representation of mux trees.
  Evaluating internal representation of mux trees.
  Analyzing evaluation results.
Removed 0 multiplexer ports.
<suppressed ~1 debug messages>

3.4. Executing OPT_REDUCE pass (consolidate $*mux and $reduce_* inputs).
  Optimizing cells in module \test.
Performed a total of 0 changes.

3.5. Executing OPT_MERGE pass (detect identical cells).
Finding identical cells in module `\test'.
Removed a total of 0 cells.

3.6. Executing OPT_DFF pass (perform DFF optimizations).
Adding EN signal on $procdff$5 ($dff) from module test (D = $add$test.sv:9$2_Y [0], Q = \count).

3.7. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \test..
Removed 1 unused cells and 3 unused wires.
<suppressed ~2 debug messages>

3.8. Executing OPT_EXPR pass (perform const folding).
Optimizing module test.

3.9. Rerunning OPT passes. (Maybe there is more to do..)

3.10. Executing OPT_MUXTREE pass (detect dead branches in mux trees).
Running muxtree optimizer on module \test..
  Creating internal representation of mux trees.
  No muxes found in this module.
Removed 0 multiplexer ports.

3.11. Executing OPT_REDUCE pass (consolidate $*mux and $reduce_* inputs).
  Optimizing cells in module \test.
Performed a total of 0 changes.

3.12. Executing OPT_MERGE pass (detect identical cells).
Finding identical cells in module `\test'.
Removed a total of 0 cells.

3.13. Executing OPT_DFF pass (perform DFF optimizations).

3.14. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \test..

3.15. Executing OPT_EXPR pass (perform const folding).
Optimizing module test.

3.16. Finished OPT passes. (There is nothing left to do.)

4. Executing JSON backend.

5. Printing statistics.

=== test ===

   Number of wires:                  4
   Number of wire bits:             35
   Number of public wires:           3
   Number of public wire bits:       3
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                  2
     $add                            1
     $dffe                           1

End of script. Logfile hash: 53f81220d2, CPU: user 0.00s system 0.00s
Yosys 0.23 (git sha1 UNKNOWN, ccache clang 11.0.0-2~ubuntu20.04.1 -Os -flto -flto)
Time spent: 4% 4x opt_expr (0 sec), 4% 3x opt_merge (0 sec), ...
`

GrootOfSDUST, 2023-01-10:
I missed the same error, but my os is ubuntu2204.
[image]

GrootOfSDUST, 2023-01-10:
I found that teros' schematic viewer didn't work properly if I used ssh to connect to Ubuntu. But if I look directly at the Schematic viewer using VScode in Ubuntu, it works fine.
Under windows-ssh
[image]

under Ubuntu
[image]

qarlosalberto, 2023-06-26:
fixed on v5.0.0